Azure/iot-edge-opc-proxy #13: Possible version missmatch in submodules

JuergenKosel, 2017-02-28:
Hello,

the .gitmodules file of this iot-gateway-proxy project references some submodules, which are already included by another submodule.
This lead for example, that 2 different versions of the libwebsocket are used now:

`$ git -C deps/lws/ describe`
`v2.0.0-202-gda6f03b`
`$ git -C deps/shared/deps/libwebsockets/ describe`
`2017_01_19-59-g7642743`

Maybe this is no problem now. But at least it has the potential of programmer confusion and problems in future.
It should be possible for the super-project to re-use the sub-modules of the sub-modules.

Greetings
  Juergen

marcschier, 2017-03-01:
Agreed, this is indeed annoying, we experimented with re-use and bringing in submodule dependencies as installable packages, etc..  You cannot do away with the grand children submodules since you would want to build it stand alone also.  In essence, we did not find a good way other than attempting to keep them all always at the same version.  

My attempt was to only require first level submodules for this project, i.e. you should not use `--recursive `when cloning, but `git submodule update --init` should just work.  Then you would not have any deeper levels of submodules.  Unfortunately once you have them all recursively pulled, there is no going back.

marcschier, 2017-03-01:
Closing, until we see an issue building / extending / consuming the project.

closed 2017-03-01 by marcschier