Azure/iot-edge-opc-proxy #14: iot-gateway-proxy needs to establish mqtt connection to port 8883

JuergenKosel, 2017-02-28:
Hello,

for the first part of session establishment, the iot-gateway-proxy tries to establish an mqtt connection to port 8883.
If this port is blocked, by some company-wide firewall rules, the iot-gateway-proxy can not connect to azure.

Greetings
  Juergen

marcschier, 2017-03-03:
Yes, this is indeed a missing feature, i.e. proxy authentication for libwebsocket.  I will add this to the backlog.

marcschier, 2017-03-03:
Failover is intentional.  However, if a configuration option would be needed, we could add that as well, e.g. always use websockets.

JuergenKosel, 2017-03-03:
This means, that intended behaviour is, that proxyd tries to establish an MQTT connection to port 8883.
And if this fails, it tries to connect through websockets?

If the user already knows, that the firewall blocks any outgoing connection, except to port 80 and 443, than this configuration option to use always websockets would save a lot of time.

Greetings  Juergen

JuergenKosel, 2017-03-03:
Regarding the proxy support on Linux:
libwebsocket should already use the proxy configuration from environment variable `http_proxy`.
But in lws_config.h the macro `LWS_WITH_HTTP_PROXY` is not defined. -> So it is disabled!
I guess, this is something we could fix in the cmake files.

In some of the libwebsocket READMEs I have read, that proxy authentication is not implemented.

Greetings
  Juergen

JuergenKosel, 2017-03-06:
Hello,

following the description of libwebsockets, proxy support could be enabled with the cmake option:
-DLWS_WITH_HTTP_PROXY:BOOL=ON

But this means, that libwebsockets needs to be linked agains libhubbub: http://www.netsurf-browser.org/projects/hubbub/ which comes with MIT license. It is available as git repository in http://source.netsurf-browser.org/libhubbub.git/ .

JuergenKosel, 2017-03-07:
Hello,

I have just seen, that an similar issue in the Azure/azure-iot-sdk-c ( https://github.com/Azure/azure-iot-sdk-c/issues/45 ) has been fixed in the libwebsockets: https://github.com/warmcat/libwebsockets/issues/764
But unfortunately the commit 71bb400, which fixes this issue is not part of the v2.1-stable branch.
And so this issue is open here.

Greetings
  Juergen

JuergenKosel, 2017-03-17:
Hello,

the issue has been _partially solved by commit 23c9424175dc50673d5682818df98539bf6b1d9a : 
Thanks to the new option --only-websocket and the environment variable http_proxy in the usual way for Linux ( e.g. `http_proxy='<user>:<password>@proxy.example.org:8080'` ) the iot-gateway-proxy successfully connects to the Azure IoT-hub, if the device-id is already registered.

But for registration and unregistration of the device-id at the Iot-hub, the iot-gateway-proxy program tries to connect directly:

`proxyd -i -n uagate-register-test -C connectstring --ns-db-file azure --only-websocket --proxy-user=<me> --proxy-pwd='<my-pass>' --proxy=proxy.example.org:8080`

This fails due to firewall restrictions, which should enforce the usage of a http_proxy:
`Error: Time:Fri Mar 17 10:39:35 2017 File:iot-gateway-proxy/deps/shared/adapters/httpapi_curl.c Func:HTTPAPI_ExecuteRequest Line:534 curl_easy_perform() failed: Couldn't connect to server`
`Error: Time:Fri Mar 17 10:39:35 2017 File:iot-gateway-proxy/deps/shared/adapters/httpapi_curl.c Func:HTTPAPI_ExecuteRequest Line:536 (result = HTTPAPI_OPEN_REQUEST_FAILED)`
`Error: Time:Fri Mar 17 10:39:35 2017 File:iot-gateway-proxy/deps/shared/src/httpapiex.c Func:HTTPAPIEX_ExecuteRequest Line:475 unable to recover sending to a working state`
`ERROR: er_connecting Install failed! Check parameters...`

And in parallel it could be seen, that the program does not try to connect to the http_proxy:
`netstat -A inet -A inet6 -np | grep proxyd`
`tcp        0      1 192.168.25.168:36320    40.118.27.192:443       SYN_SENT    18027/proxyd`

Greetings
Juergen

JuergenKosel, 2017-03-17:
For Windows I got a similar result for the registration:

`./proxyd.exe  -c <secret>  -i -n "win10-test" --only-websocket --ns-db-file azure  --proxy 'proxy.example.com:8080' --proxy-user <me> --proxy-pwd '<secret>'`
`=== Azure IoT Proxy Gateway ===`
`Error: Time:Fri Mar 17 14:46:31 2017 File:C:\Programming\iot-gateway-proxy\deps\shared\adapters\httpapi_winhttp.c Func:HTTPAPI_ExecuteRequest Line:434 WinHttpSendRequest: (result = HTTPAPI_SEND_REQUEST_FAILED).`
`Error: Time:Fri Mar 17 14:46:31 2017 File:C:\Programming\iot-gateway-proxy\deps\shared\adapters\httpapi_winhttp.c Func:HTTPAPI_ExecuteRequest Line:434 GetLastError: Das Zeitlimit fuer den Vorgang wurde erreicht.`
.
`Error: Time:Fri Mar 17 14:46:31 2017 File:C:\Programming\iot-gateway-proxy\deps\shared\src\httpapiex.c Func:HTTPAPIEX_ExecuteRequest Line:475 unable to recover sending to a working state`
`ERROR: er_connecting Install failed! Check parameters...`

marcschier, 2017-03-17:
Yes, I need to fix that as well.  Http passes through azure shared utility http io and that needs proxy settings passed as option setting.  I will keep this open and address it next week

JuergenKosel, 2017-03-29:
Hello @marcschier,

my colleague has already discovered, that this still does not work. Because not the proxy host, provided in the option argument of --proxy is used, instead something like "/etc/ssl" is used as proxy host.
I think, she has already found were it happens. So we will work on another pull request to fix this.

Greetings
Juergen

marcschier, 2017-03-29:
Hi @JuergenKosel, thanks a bunch and thanks for the help and support!

JuergenKosel, 2017-03-29:
Hello,

the debugger shows in function HTTPAPI_CloneOption file iot-gateway-proxy/deps/shared/adapters/httpapi_curl.c:824 the host_address is copied correctly.
But in HTTPAPI_SetOption file iot-gateway-proxy/deps/shared/adapters/httpapi_curl.c, line 736, the proxy_data->host_address has been overwritten.
So this could be an issue of the shared submodule!

JuergenKosel, 2017-03-29:
Hello,

we have found the problem in prx_ns_iot_hub_rest_call_configure_proxy file src/prx_ns.c:1056

```
    // Create a copy so we can parse the port number
    result = string_clone(ptr, &buffer);
    if (result != er_ok)
        return;
    proxy_options.host_address = ptr = buffer;
    proxy_options.port = 0;
    while (*ptr && *ptr != ':')
        ptr++;
    if (*ptr)
    {
        *(char*)ptr = 0;
        proxy_options.port = atoi(++ptr);
    }
    if (HTTPAPIEX_OK != HTTPAPIEX_SetOption(http_handle, OPTION_HTTP_PROXY, 
        &proxy_options))
    {
        log_debug(NULL, "Failed to configure proxy settings with httpapi");
    }
    mem_free(buffer);
```
With the call of HTTPAPIEX_SetOption the pointer buffer becomes part of the http_handle structure.
And later on, the memory of buffer is released.
Since the next malloc assigns the same memory to another string, our application does not crash with a segmentation fault. But the wrong string is used!
So the mem_free() needs to be removed here!

JuergenKosel, 2017-03-29:
As already reported in #25 there is a memory corruption, related to the --proxy option, which could be reproduces on armhf-linux and amd64-linux:

```
Starting program: /tmp/azure/usr/local/bin/proxyd -D iot-gateway-proxy/azure --proxy proxy.example.com  --proxy-user me --proxy-pwd 'secret' --only-websocket
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7ffff313c700 (LWP 5490)]
[New Thread 0x7ffff28eb700 (LWP 5491)]
(gdb) bt
#0  malloc_consolidate (av=av@entry=0x7fffec000020) at malloc.c:4157
#1  0x00007ffff77c3e98 in _int_malloc (av=0x7fffec000020, bytes=5008) at malloc.c:3423
#2  0x00007ffff77c6020 in __GI___libc_malloc (bytes=5008) at malloc.c:2891
#3  0x00007ffff6d99519 in _realloc (ptr=0x0, size=5008) at iot-gateway-proxy/deps/lws/lib/alloc.c:6
#4  0x00007ffff6d9957f in lws_zalloc (size=5008) at iot-gateway-proxy/deps/lws/lib/alloc.c:21
#5  0x00007ffff6d98472 in lws_create_context (info=0x7ffff28eaaa0) at iot-gateway-proxy/deps/lws/lib/context.c:602
#6  0x00007ffff7b60508 in pal_wsworker_pool_attach (pool=0x638c50, wsclient=0x7fffec001100) at /home/klj/iot-gateway-proxy/src/pal/pal_ws_lws.c:1174
#7  0x00007ffff7b60bc6 in pal_wsclient_connect (wsclient=0x7fffec001100) at iot-gateway-proxy/src/pal/pal_ws_lws.c:1378
#8  0x00007ffff7b56934 in xio_wsclient_open (handle=0x7fffec0009e0, on_io_open_complete=0x7ffff7ba601d <onOpenComplete>, on_io_open_complete_context=0x7fffec0008c0, on_bytes_received=0x7ffff7ba6233 <onBytesReceived>, on_bytes_received_context=0x7fffec0008c0, on_io_error=0x7ffff7ba62cb <onIoError>, on_io_error_context=0x7fffec0008c0) at iot-gateway-proxy/src/xio_ws.c:482
#9  0x00007ffff7b754d1 in xio_open (xio=0x7fffec0009c0, on_io_open_complete=0x7ffff7ba601d <onOpenComplete>, on_io_open_complete_context=0x7fffec0008c0, on_bytes_received=0x7ffff7ba6233 <onBytesReceived>, on_bytes_received_context=0x7fffec0008c0, on_io_error=0x7ffff7ba62cb <onIoError>, on_io_error_context=0x7fffec0008c0) at /home/klj/iot-gateway-proxy/deps/shared/src/xio.c:88
#10 0x00007ffff7ba796b in mqtt_client_connect (handle=0x7fffec0008c0, xioHandle=0x7fffec0009c0, mqttOptions=0x7ffff28eadb0) at iot-gateway-proxy/deps/umqtt/src/mqtt_client.c:896
#11 0x00007ffff7b3fd3c in io_mqtt_connection_reconnect (connection=0x652a00) at /home/klj/iot-gateway-proxy/src/io_mqtt.c:1036
#12 0x00007ffff7b2efb3 in prx_scheduler_work (context=0x63a710) at iot-gateway-proxy/src/prx_sched.c:229
#13 0x00007ffff7b791e2 in ThreadWrapper (threadInstanceArg=0x63a330) at /home/klj/iot-gateway-proxy/deps/shared/adapters/threadapi_pthreads.c:34
#14 0x00007ffff72c1064 in start_thread (arg=0x7ffff28eb700) at pthread_create.c:309
#15 0x00007ffff783262d in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:111

```

JuergenKosel, 2017-03-30:
Pull request #28 fixes the segmentation fault, but using the --proxy option works still only for registration and unregistration:

`Error: Time:Thu Mar 30 09:58:14 2017 File:iot-gateway-proxy/src/pal/pal_ws_lws.c Func:pal_wsclient_lws_log Line:115 ERROR proxy: HTTP/1.1 407
`
If the environment variable http_proxy is used, the connection over the proxy works fine.

marcschier, 2017-08-14:
Closing since there is a work around.  Please open feature request if we need to improve this area.

closed 2017-08-14 by marcschier