Azure/iot-edge-opc-proxy #15: More detailed description of (sample) usage needed

JuergenKosel, 2017-03-01:
Hello,

could you please describe how to start the proxyd to run the samples, or to connect the sample opc ua client to another opc ua server.
If I start the proxyd with no aruments, it just prints:
"=== Azure IoT Proxy Gateway ===" end terminates.
So I guess, some options needs to be set first.

Greetings
  Juergen

JuergenKosel, 2017-03-02:
Hello,

please also add Hyperlinks to the related azure documentation, e.g. https://docs.microsoft.com/de-de/azure/service-bus-relay/relay-create-namespace-portal for the creation of the mentioned "Service Bus Hybrid Connection Namespace".
Otherwise the user needs to search in the online help and hope that there are not to many Topics which accidently also match the search pattern.

Greetings
Juergen

marcschier, 2019-07-02:
Removed service bus relay support, thus closing this documentation issue.

closed 2019-07-02 by marcschier