Azure/iot-edge-opc-proxy #16: The current working directory is prepended to the ns-db-file, even if the user has already provided a full path

JuergenKosel, 2017-03-02:
Hello,

with the command line option `-d <file-name>` or `--ns-db-file <file-name>` could provide the user the file to store local name service database.
If the user provides an absolute path as argument for this option (e.g. because the proxyd should be started from some script), then the current working directory is prep-ended to the absolute path.
So that proxyd fails to open the given file.
I recommend to either document this behavior. Or to modify function pal_create_full_path() to return
file_name, if it is already an absolute path.

Greetings
  Juergen

marcschier, 2017-03-02:
This will be fixed soon.

Outlook for Android<https://aka.ms/ghei36> herunterladen

JuergenKosel, 2017-03-10:
Fixed in 0e4d9afe

closed 2017-03-10 by JuergenKosel