Azure/iot-edge-opc-proxy #17: proxyd -C <filename> does not read the connect string from file

JuergenKosel, 2017-03-03:
Hello,

since the connect string contains characters which have special meaning for the shell,
I want to use the -C option to read in the connection string into proxyd.
But this option does not work as described:
Function `prx_host_init_from_command_line()` calls `prx_ns_iot_hub_create()` if option "-C" is used.
`prx_ns_iot_hub_create()` calls `io_cs_create_from_string(config, &cs)`.
Which is the same as `prx_host_init_from_command_line()` does for the "-c" option.
So the argument of -C is not interpreted as a filename but as connection string!

Greetings Juergen

JuergenKosel, 2017-03-03:
Hello,

for Linux (and the git bash, which is installed with Visual Studio) the following workaround exists:

`read -r _HUB_CS < connectstring_file`
`export _HUB_CS`
Then proxyd reads the connect string from the environment variable _HUB_CS.

Greetings  Juergen

marcschier, 2017-03-03:
This will be fixed, then -C will read from raw file rather than from a JSON encoded representation of hub entries.  Agreed that the fact that one cannot parse the string as a connection string, does not mean it is a file name, I fixed that also.

JuergenKosel, 2017-03-08:
Fixed in commit 0e4d9afe

closed 2017-03-08 by JuergenKosel