Azure/iot-edge-opc-proxy #23: Fail to run OPC UA sample

JuergenKosel, 2017-03-23:
Hello,

while trying to follow the description in https://github.com/Azure/iot-gateway-opc-ua-proxy/blob/master/api/csharp/samples/opc-ua/readme.md
the sample Client fails to connect to the sample Server application. While the proxyd is running and successfully connected to the IoT-hub (as visible in DeviceExplorer):

```
C:\Programmining\iot-gateway-proxy\api\csharp\samples\opc-ua\test\Client>bin\Debug\net46\Client.exe
.Net Core OPC UA Console Client sample
1 - Create an Application Configuration.
2 - Discover endpoints of opc.tcp://desktop-g0fuqb8:51210/UA/SampleServer.
Could not fetch endpoints from url: opc.tcp://desktop-g0fuqb8:51210/UA/SampleServer
Reason = Error establishing a connection: BadNotConnected
Exit due to Exception: Mindestens ein Fehler ist aufgetreten.

```

It looks like, that the TCP Connection could not be established. So that failure is Independent frm OPC UA SSL/TLS certificates.

Greetings
Juergen

marcschier, 2017-03-24:
Indeed that is what it looks like.  Firstly, are you setting _HUB_CS environment variable before running the sample?  This looks like there is an issue just connecting to IoTHub, and connection string is usually the cause of that.   Otherwise there would be a timeout error most likely.  

When you run the simple.exe sample, does that work?  If so, do you see anything in the Diagnostic Trace that does not look right?  Also, is there activity in the proxy console during connect?

marcschier, 2017-03-24:
Just noticed that there is no reference to _HUB_CS var in the md you linked.  That should be fixed.

JuergenKosel, 2017-03-24:
With _HUB_CS set, it is even more worse, the program "crashes":

```
C:\Programmining\iot-gateway-proxy\api\csharp\samples\opc-ua\test\Client>bin\Debug\net46\Client.exe
.Net Core OPC UA Console Client sample
1 - Create an Application Configuration.
2 - Discover endpoints of opc.tcp://desktop-g0fuqb8:51210/UA/SampleServer.

Unbehandelte Ausnahme: System.IO.FileLoadException: Die Datei oder Assembly "Microsoft.Win32.Primitives, Version=4.0.0.0, Culture=neutral, PublicKeyToken=b03f5f7f11d50a3a" oder eine Abhängigkeit davon wurde nicht gefunden. Die gefundene Manifestdefinition der Assembly stimmt nicht mit dem Assemblyverweis überein. (Ausnahme von HRESULT: 0x80131040)
   bei System.Net.Http.WinHttpHandler.HandleAsyncException(WinHttpRequestState state, Exception ex)
   bei System.Net.Http.WinHttpHandler.<StartRequest>d__101.MoveNext()
--- Ende der Stapelüberwachung vom vorhergehenden Ort, an dem die Ausnahme ausgelöst wurde ---
   bei System.Runtime.CompilerServices.AsyncMethodBuilderCore.<>c.<ThrowAsync>b__6_1(Object state)
   bei System.Threading.QueueUserWorkItemCallback.WaitCallback_Context(Object state)
   bei System.Threading.ExecutionContext.RunInternal(ExecutionContext executionContext, ContextCallback callback, Object state, Boolean preserveSyncCtx)
   bei System.Threading.ExecutionContext.Run(ExecutionContext executionContext, ContextCallback callback, Object state, Boolean preserveSyncCtx)
   bei System.Threading.QueueUserWorkItemCallback.System.Threading.IThreadPoolWorkItem.ExecuteWorkItem()
   bei System.Threading.ThreadPoolWorkQueue.Dispatch()
   bei System.Threading._ThreadPoolWaitCallback.PerformWaitCallback()

```

marcschier, 2017-03-24:
Ah, but that is better ;).  What happens when you run this in VS? 

There are also 2 exe built by VS, make sure you run the one in the subfolder which has all dependencies unpacked.  If that fails, please try the net core one to narrow it down.  You can run this by calling 'dotnet.exe client.dll'.

JuergenKosel, 2017-03-24:
The variant in the net46 directory is the one which seems to have all dependencies unpacked.
For the other 2 directories, there is no client.exe and some warning appeared during build for them.

marcschier, 2017-03-24:
The warning is a .Net core 1.1.1 artifact coming from the relay assembly but it is benign, and should be gone when relay releases new nuget packages.  

.Net core itself does not know .Exe files.  All apps are .Dll images that are executed/hosted by dotnet executable.  So in this case you run 'dotnet client.dll' similar to 'java jar-file'.  Did the subfolder client.exe work for you?

JuergenKosel, 2017-03-27:
Hello,

the .exe created the certificate. But during the Connection attempt it crashed (as above).

The dotnet .dll works now, if started from the correct working directory:

`C:\Programmining\iot-gateway-proxy\api\csharp\samples\opc-ua\test\Client>dotnet bin\Debug\netcoreapp1.1\Client.dll opc.tcp://uagate0012.inat.internal:4840/`

Greeting
Juergen

closed 2017-03-27 by JuergenKosel

murugend, 2017-06-01:
Hi

I don't completely understand how this issue was resolved. What is the _HUB_CS and where is it set. Is this the connection String to the IoTHub or the the connection string to the device (the proxy)?

Thanks
Boathy

JuergenKosel, 2017-06-01:
Hello,

you need to set the _HUB_CS variable similar to the setting of the PATH environment variable
In a .bat script or in the command.com terminal enter:
`set _HUB_CS=HostName=my-iot-hobname....`

murugend, 2017-06-01:
Thanks @JuergenKosel , all working now:)