Azure/iot-edge-opc-proxy #26: Registration and unregistration occassionally fail

JuergenKosel, 2017-03-29:
Sometimes the first attempt to register a device-id at the IoT-hub fails in the first attempt:

```
=== Azure IoT Proxy Gateway === 
Version: 0.1.1.0.1.1-60-gd6e1962
Error: Time:Wed Mar 29 13:24:01 2017 iot-gateway-proxy/deps/shared/adapters/httpapi_curl.c Func:HTTPAPI_ExecuteRequest Line:594 Failure in HTTP communication: server reply code is 500
Info: HTTP Response:{"Message":"ErrorCode:ServerError;InternalServerError","ExceptionMessage":"Tracking ID:65d1a462f772409fb0452c4ff17f0a8b-G:5-TimeStamp:03/29/2017 13:24:03"}
ERROR: er_fatal Install failed! Check parameters...
Operation failed.

```

Another attempt a few seconds later succeeds.

JuergenKosel, 2017-04-06:
Sometimes the registration fails completely with an other error output:

```
=== Azure IoT Proxy Gateway === 
Error: Time:Mon Apr  3 09:30:14 2017 File:iot-gateway-proxy/deps/shared/adapters/httpapi_curl.c Func:HTTPAPI_ExecuteRequest Line:594 Failure in HTTP communication: server reply code is 400
Info: HTTP Response:{"Message":"ErrorCode:ArgumentNull;BadRequest","ExceptionMessage":"Tracking ID:3f2bd978074a49868e35d0c8c3f259d9-G:12-TimeStamp:04/03/2017 09:30:13"}
ERROR: er_invalid_format Install failed! Check parameters...
Operation failed. (Step 6)

```

And then another attempt with the same parameters succeeds.

marcschier, 2017-06-13:
Please log this issue through the portal once it happens again.  This is the best path to follow through on this unfortunately.

closed 2017-06-13 by marcschier