Azure/iot-edge-opc-proxy #37: Bugs in OPC UA Test Client

faister, 2017-05-21:
I tried to run the [OPC UA Test Client ](https://github.com/Azure/iot-edge-opc-proxy/blob/master/api/csharp/samples/opc-ua/test/Client/Program.cs ) and exception was thrown. I debugged the console app and it's full of bugs.

1. Failed at discovering endpoints from the OPC UA Server. 
2. Could not create session with OPC UA Server
3. When trying to access the underlying proxy socket, the object is null. this is from this [line](https://github.com/Azure/iot-edge-opc-proxy/blob/master/api/csharp/samples/opc-ua/test/Client/Program.cs#L149)

For (1) and (2) I replaced the code with code from [NetCoreConsoleClient](https://github.com/OPCFoundation/UA-.NETStandardLibrary/blob/master/SampleApplications/Samples/NetCoreConsoleClient/Program.cs)

I don't know how the proxy socket works, hence unable to fix it. Need better explanation in the documentation about how connection from the OPA UA Test client to the OPC UA Server is linked through the OPC UA Proxy.

marcschier, 2017-05-21:
Hi @faister.  Thanks for digging in and debugging.  Agreed, better docs are a theme of the issues list.  If you have suggestions, e.g. questions I can better answer in the existing docs, let me know.  

To run the client 2 things are required.  1. The _HUB_CS environment variable must be set as the iothubowner connection string, 2. A proxy must be running in the same network as the OPC server you are trying to access.  

What the samples should do when no connection string is found is to bail out with a helpful exception message.  I will add this to our backlog.

If you found (other) issues while debugging, please do not hesitate to send a PR with potential fixes.  Thx!

marcschier, 2017-05-24:
Hi @faister. Since I did not hear from you, please re-open if above did not yield results, or if you have a more concrete issue / repro step you can share.  Thx :-)

closed 2017-05-24 by marcschier