Azure/iot-edge-opc-proxy #39: Optionally limit the destination addresses and port numbers of the proxyd application

JuergenKosel, 2017-05-31:
As depicted in
https://github.com/Azure/iot-edge-opc-proxy/tree/0.2.0/api/csharp/samples/netcat
the proxyd application allows to tunnel each TCP based protocol.

For security reasons it might be necessary to limit the addresses and port numbers, which could be addressed by the proxyd application.
For example it should be possible to limit proxyd to allow connections only to one specific OPC UA server with port range 4840 to 4842:
proxyd --destination-host <allowed-server-hostname> --destination-port 4840:4842

marcschier, 2017-05-31:
Yes, i.e. limit host addresses to the subnet and limit port ranges to avoid configuration of firewalls, we discussed this as well.  I'm going to add this to my backlog.

marcschier, 2017-06-13:
Limit using --blacklisted-ports, (reverse than whitelist) in #46.

marcschier, 2017-07-27:
The feature that will be added in 1.0.3 is a way to pass an interface name (e.g. eth0) to confine all access on this socket to this interface.  I.e. this amounts to setting SO_BIND_TO_DEVICE on Linux.  However, since the latter requires appropriate CAPS, it might fail should the context not have them.  The current plan then would be to continue with the connection.  This is aligned with other OSS projects' use of the BIND_TO_DEVICE.  

Any further restriction on host/IP would be something we would happily take as project specific contribution.

JuergenKosel, 2017-08-04:
Hello,

it looks like, there is a problem with the blacklisted ports and the IoT-Hub:
We have several devices, each running an instance of proxyd and an opc ua server.
Where each proxyd allows connections only to the port number which matches the port of the own opc ua server.
Sometimes the IoT-hob (connected factory) can not connect to an opc ua server, which should be reachable over at least one proxyd instance.
I have the impression, that the IoT hub tries to connect through a proxyd instance, which has the port of the foreign opc ua server blacklisted, or even more worse, is in the wrong network.

Are there some means, that the proxyd could report to the IoT hub, that a port is blacklisted?
So that the IoT hub could try to connect through another proxyd?

marcschier, 2017-08-04:
Right now a ping_request that is blocked by a blacklisted port returns er_refused.  Since otherwise getaddrinfo is used which should not generate the same error code one could distinguish a blacklisted port from a not found address I imagine.  

To diagnose it would be good to get a bit more detail re: the setup.  I.e. network topology, proxies, opc server names, host names.  Also, are you able to use the opc command line test client to repro the same?  That way one could enable more traces to see where exactly the issue occurs.

Also, if ports are not black listed, I assume you do not see this issue?  

I am breaking this issue off from the feature request so we can track both independently.

marcschier, 2017-08-04:
Btw, a good place to put a breakpoint is in ProxySocket.cs, line 213 where the ping responses roll back.

marcschier, 2017-08-14:
Please reopen with more concrete info so we can reproduce this on our end and triage.  Thx :-)

closed 2017-08-14 by marcschier