Azure/iot-edge-opc-proxy #43: proxyd can not connect to Azure IoT-hub anymore

JuergenKosel, 2017-06-07:
Hello,

until yesterday the proxyd could connect to my Azure IoT-hub.
But now it fails. But Registration is still working:

```
D:\Programmierung\iot-gateway-opc-ua-proxy\build\cmake\x64\bin\Debug>proxyd -i -n test-laptop -D azure.json

=== Azure iot-edge-reverse-proxy 0.2.0.0.1.4-revised-4-gdccda9d ===

test-laptop installed
Success.

D:\Programmierung\iot-gateway-opc-ua-proxy\build\cmake\x64\bin\Debug>proxyd -D azure.json

=== Azure iot-edge-reverse-proxy 0.2.0.0.1.4-revised-4-gdccda9d ===

Wed Jun  7 09:27:33 2017 (prx_host_init:1024) [DEBUG] Proxy host created!
Wed Jun  7 09:27:33 2017 (pal_getaddrinfo:988) [INFO] Resolving IoT-Hub.devices.net:8883 (family: 0)...
Wed Jun  7 09:27:33 2017 (pal_socket_async_connect_complete:462) [DEBUG] Socket connected asynchronously!
Wed Jun  7 09:27:33 2017 (io_mqtt_connection_handle_CONNECT_ACK:707) [INFO] Connection established (IoT-Hub.azure-devices.net:8883)!
Wed Jun  7 09:27:40 2017 (pal_sdclient_create:715) [ERROR] Failed creating connection to service: -65563 (er_bad_state)
Wed Jun  7 09:27:40 2017 (io_mqtt_connection_close:1478) [DEBUG] Closing connection ...
Wed Jun  7 09:27:40 2017 (io_mqtt_subscription_free:1129) [DEBUG] Free subscription for topic $iothub/methods/POST/#...
Wed Jun  7 09:27:40 2017 (io_mqtt_connection_close:1493) [DEBUG] Begin disconnect and schedule free ...
Wed Jun  7 09:27:40 2017 (io_iot_hub_umqtt_connection_on_free:134) [INFO] mqtt transport connection closed.
Wed Jun  7 09:27:40 2017 (prx_module_create:113) [ERROR] Failed to create server (er_bad_state)!
Wed Jun  7 09:27:40 2017 (tlsio_schannel_close:1175) [ERROR] invalid tls_io_instance->tlsio_state = TLSIO_STATE_CLOSING
Wed Jun  7 09:27:40 2017 (tlsio_schannel_close:1175) [ERROR] invalid tls_io_instance->tlsio_state = TLSIO_STATE_CLOSING
Wed Jun  7 09:27:40 2017 (io_mqtt_connection_complete_disconnect:896) [DEBUG] Socket disconnected.
Wed Jun  7 09:27:40 2017 (io_mqtt_connection_complete_disconnect:904) [DEBUG] Client disconnected.
Wed Jun  7 09:27:40 2017 (io_mqtt_connection_free:241) [DEBUG] Connection freed.
Wed Jun  7 09:27:40 2017 (prx_server_free:197) [DEBUG] Freeing server.
Proxy exits... Goodbye!

```

JuergenKosel, 2017-06-07:
Just checked with an earlier Version (0.1.4-revised). With this Version the Connection could be established:

```
D:\Programmierung\iot-gateway-opc-ua-proxy\build\cmake\x64\bin\Debug>proxyd -i -n test-laptop -D azure.json
=== iot-edge-reverse-proxy 0.1.4.develop ===
test-laptop installed
Success.

D:\Programmierung\iot-gateway-opc-ua-proxy\build\cmake\x64\bin\Debug>proxyd -D azure.json -v
=== iot-edge-reverse-proxy 0.1.4.develop ===
Version: 0.1.4.0.1.4-revised
Wed Jun  7 11:33:17 2017 (prx_host_init:1014) [DEBUG] Proxy host created!
Wed Jun  7 11:33:17 2017 (prx_module_create:117) [DEBUG] Proxy module created in GW Host!
Wed Jun  7 11:33:17 2017 (pal_socket_open_by_name_begin:1192) [INFO] Resolving IoT-Hub.azure-devices.net:8883...
Wed Jun  7 11:33:17 2017 (prx_host_start:792) [INFO] Host server started!
Wed Jun  7 11:33:17 2017 (pal_getaddrinfo:945) [INFO] getaddrinfo for "IoT-Hub.azure-devices.net", "8883" family: 0
Wed Jun  7 11:33:17 2017 (pal_socket_async_connect_complete:381) [DEBUG] Socket connected asynchronously!
Wed Jun  7 11:33:18 2017 (io_mqtt_connection_handle_CONNECT_ACK:705) [INFO] Connection established (IoT-Hub.azure-devices.net:8883)!
Wed Jun  7 11:33:28 2017 (io_mqtt_connection_operation_callback:794) [INFO] Connection alive...
Wed Jun  7 11:33:38 2017 (io_mqtt_connection_operation_callback:794) [INFO] Connection alive...
Wed Jun  7 11:33:48 2017 (io_mqtt_connection_operation_callback:794) [INFO] Connection alive...
Wed Jun  7 11:33:58 2017 (io_mqtt_connection_operation_callback:794) [INFO] Connection alive...
```

This means this is a Regression issue!

JuergenKosel, 2017-06-07:
Hello,

for some reason I can not reproduce this issue on my usual Linux PC:
The proxyd build on my PC from current sources, works fine. If I cross-compile the proxyd for linux-armhf, it works fine.
But if I cross-compile the proxyd on some other PC it builds without error or warning. But the proxyd can not connect to the Azure IoT-hub. (Except for registration).
Also the proxyd build in Visusal Studio 2017 does not work (as reported in the first message).

Maybe this issue is caused by some bad order in the CMakeList.txt files...

marcschier, 2017-06-08:
The idea is that the stub just does not enable browsing in a proxy instance, but it should not sigill.  I.e. there needs to be a check before create is called whether sd cap is set.  This is a bug, which needs to be fixed.

marcschier, 2017-06-09:
#46 should fix this.

marcschier, 2017-06-13:
Fixed in #46

closed 2017-06-13 by marcschier