Azure/iot-edge-opc-proxy #47: Opc Ua Sample: Random disconnects

PedroFloresGuillen, 2017-06-12:
While doing the OPC UA Sample (/samples/opc-ua/readme.md), the client will start to fecth the server info, all good so far, but then, randomly it will stop fecthing , while the proxy gives:

Mon Jun 12 15:49:28 2017 (io_iot_hub_umqtt_connection_on_receive:438) [INFO] IN: [LINK]
Mon Jun 12 15:49:28 2017 (pal_getaddrinfo:988) [INFO] Resolving desktop-na6lmmm:51210 (family: 2)...
Mon Jun 12 15:49:28 2017 (pal_socket_async_connect_complete:462) [DEBUG] Socket connected asynchronously!
Mon Jun 12 15:49:28 2017 (io_iot_hub_umqtt_connection_on_send:316) [INFO] OUT: [LINK]

and the client:
Exit due to Exception: One or more errors occurred. (BadSecureChannelClosed)

I'm quite lost here, since this error can appear as soon as 1 min and as late (as recorded) as 7 min, while fetching different nodes.

marcschier, 2017-06-12:
Hi, could you provide more info re. your setup, i.e. the server you are connecting to so we can repro?  There is an open issue in the stack that manifests quicker with the proxy

PedroFloresGuillen, 2017-06-13:
Window 10, visual studio 2017, server is the sample from UA-.NETStandardLibrary and the client is the client that comes with the proxy

marcschier, 2017-07-21:
Please try release 1.0.2 to see if you can still repro this.

PedroFloresGuillen, 2017-07-21:
[image]

downloaded, builded and tested, got a stack overflow now

marcschier, 2017-07-21:
Yes, this is tracked in issue #55.  Can you try to delete the OPC Foundation folder?  Do you get the stack overflow then?

[excpetion]<https://na01.safelinks.protection.outlook.com/?url=https%3A%2F%2Fuser-images.githubusercontent.com%2F28599539%2F28458963-e1130ca6-6e0b-11e7-89d0-852c3d40e0a8.png&data=02%7C01%7CMarc.Schier%40microsoft.com%7C7ebd1989e5a748d9871808d4d01f2a37%7C72f988bf86f141af91ab2d7cd011db47%7C1%7C0%7C636362279647948300&sdata=oWxGJsUXw2zxnmlF7%2FGfjJcMPa3F9apLfKZkBUOatQI%3D&reserved=0>

PedroFloresGuillen, 2017-07-21:
Now  the proxy starts then after a while crashes

marcschier, 2017-07-21:
Could you provide a couple more data points to help us debug the crash, i.e. what is the log output at the time of the crash and what was happening on the client side?  Do you have stack traces, etc, or potentially a core dump?  I assume you are running the proxy directly on Windows 10 as originally mentioned?

marcschier, 2017-07-27:
Closing stale issue.  Please open a new bug for any crashes, ideally with the above information.

If you see BadSequenceIdInvalid errors in the .net classic server when communicating through the proxy, please reopen this issue.  There are issues with the .net Std server, which are still being investigated.

closed 2017-07-27 by marcschier