Azure/iot-edge-opc-proxy #55: OPC UA client sample can not connect

JuergenKosel, 2017-07-17:
Hello,

while running the proxyd with commit id 254b2e328026170d2316c258587f7d0dfeca30bc , the opc ua sample Client can not connect to the opc ua Servers behind the proxyd.

At the same time, the connted factory, running the Proxy_1_0_0 branch Version can connect to the opc ua Servers behind the proxyd. So I guess, that this is a problem of the sample Client.

```
D:\Programmierung\iot-gateway-opc-ua-proxy\bld>cd D:\Programmierung\iot-gateway-opc-ua-proxy\api\csharp\samples\opc-ua\test\Client

D:\Programmierung\iot-gateway-opc-ua-proxy\api\csharp\samples\opc-ua\test\Client>dotnet bin\Debug\netcoreapp1.1\Client.dll opc.tcp://[fd00:42:2001:db8:206:71ff:fe33:12]:4840/
.Net Core OPC UA Console Client sample
1 - Create an Application Configuration.
2 - Discover endpoints of opc.tcp://[fd00:42:2001:db8:206:71ff:fe33:12]:4840/.
Could not fetch endpoints from url: opc.tcp://[fd00:42:2001:db8:206:71ff:fe33:12]:4840/
Reason = Error establishing a connection: BadNotConnected
Exit due to Exception: One or more errors occurred. (Error establishing a connection: BadNotConnected)
```

With older versions of the proxyd and the corresponding sample Client, this is still successfull. (But then the proxyd is incompatible to the connected factory.)

marcschier, 2017-07-17:
Yes, I think the #if !PERF checks should be #if PERF.  That is a bug that is fixed in 1.0.1. I would think the stack overflow is from the browse recursion, but that should not occur when only the root is browsed as in 1.0.1 PR.

JuergenKosel, 2017-07-18:
I have reduced the name space of the opc ua server to just a few variabels. - And there is still the stack over flow. There must be some thing wrong in the termination of the browse recursion.

marcschier, 2017-07-18:
I will rewrite this using a stack.  Still surprised what ends up on the stack there.

This could be coming with the 0.2.4 nugets.  I think the old proxy was still on 0.2.3.

Again, non recursive is the default on the netconsoleclient.  Do you also see the exception when the client only browses the root node, i.e. the 1.0.1 behavior?

JuergenKosel, 2017-07-19:
Hello,

the stack over run exception occurs with Version 1.0.1. With the earlier Version it simply failed to connect.
Maybe this issue is related to this nuget warnings:

 ```
 Microsoft.Azure.Devices.Proxy -> D:\Programmierung\iot-gateway-opc-ua-proxy\api\csharp\src\Microsoft.Azure.Devices.Proxy\bin\Debug\netstandard1.3\Microsoft.Azure.Devices.Proxy.dll
C:\Program Files\dotnet\sdk\1.0.4\Microsoft.Common.CurrentVersion.targets(1964,5): warning MSB3277: Found conflicts between different versions of the same dependent assembly that could not be resolved.  These reference conflicts are listed in the build log when log verbosity is set to detailed. [D:\Programmierung\iot-gateway-opc-ua-proxy\api\csharp\samples\provider\relay\src\Microsoft.Azure.Devices.Proxy.Relay\Microsoft.Azure.Devices.Proxy.Relay.csproj]
  Microsoft.Azure.Devices.Proxy.Relay -> D:\Programmierung\iot-gateway-opc-ua-proxy\api\csharp\samples\provider\relay\src\Microsoft.Azure.Devices.Proxy.Relay\bin\Debug\netstandard1.3\Microsoft.Azure.Devices.Proxy.Relay.dll
  Microsoft.Azure.Devices.Proxy.AspNet -> D:\Programmierung\iot-gateway-opc-ua-proxy\api\csharp\samples\provider\webapp\src\Microsoft.Azure.Devices.Proxy.AspNet\bin\Debug\netstandard1.3\Microsoft.Azure.Devices.Proxy.AspNet.dll
  Microsoft.Azure.Devices.Proxy.Opc.Ua -> D:\Programmierung\iot-gateway-opc-ua-proxy\api\csharp\samples\opc-ua\src\Microsoft.Azure.Devices.Proxy.Opc.Ua\bin\Debug\netstandard1.3\Microsoft.Azure.Devices.Proxy.Opc.Ua.dll
  Microsoft.Azure.Devices.Proxy.Kestrel -> D:\Programmierung\iot-gateway-opc-ua-proxy\api\csharp\samples\provider\webapp\test\Host\bin\Debug\netstandard1.3\Microsoft.Azure.Devices.Proxy.Kestrel.dll
C:\Program Files\dotnet\sdk\1.0.4\Microsoft.Common.CurrentVersion.targets(1964,5): warning MSB3277: Found conflicts between different versions of the same dependent assembly that could not be resolved.  These reference conflicts are listed in the build log when log verbosity is set to detailed. [D:\Programmierung\iot-gateway-opc-ua-proxy\api\csharp\samples\opc-ua\test\Client\Client.csproj]
  Client -> D:\Programmierung\iot-gateway-opc-ua-proxy\api\csharp\samples\opc-ua\test\Client\bin\Debug\netcoreapp1.1\Client.dll
  Microsoft.Azure.Devices.Proxy -> D:\Programmierung\iot-gateway-opc-ua-proxy\api\csharp\src\Microsoft.Azure.Devices.Proxy\bin\Debug\net46\Microsoft.Azure.Devices.Proxy.dll
  Microsoft.Azure.Devices.Proxy.Relay -> D:\Programmierung\iot-gateway-opc-ua-proxy\api\csharp\samples\provider\relay\src\Microsoft.Azure.Devices.Proxy.Relay\bin\Debug\net46\Microsoft.Azure.Devices.Proxy.Relay.dll
  Microsoft.Azure.Devices.Proxy.AspNet -> D:\Programmierung\iot-gateway-opc-ua-proxy\api\csharp\samples\provider\webapp\src\Microsoft.Azure.Devices.Proxy.AspNet\bin\Debug\net46\Microsoft.Azure.Devices.Proxy.AspNet.dll
  Microsoft.Azure.Devices.Proxy.Opc.Ua -> D:\Programmierung\iot-gateway-opc-ua-proxy\api\csharp\samples\opc-ua\src\Microsoft.Azure.Devices.Proxy.Opc.Ua\bin\Debug\net46\Microsoft.Azure.Devices.Proxy.Opc.Ua.dll
  Microsoft.Azure.Devices.Proxy.Kestrel -> D:\Programmierung\iot-gateway-opc-ua-proxy\api\csharp\samples\provider\webapp\test\Host\bin\Debug\net46\Microsoft.Azure.Devices.Proxy.Kestrel.dll
  Client -> D:\Programmierung\iot-gateway-opc-ua-proxy\api\csharp\samples\opc-ua\test\Client\bin\Debug\net46\Client.exe

Build succeeded.

C:\Program Files\dotnet\sdk\1.0.4\Microsoft.Common.CurrentVersion.targets(1964,5): warning MSB3277: Found conflicts between different versions of the same dependent assembly that could not be resolved.  These reference conflicts are listed in the build log when log verbosity is set to detailed. [D:\Programmierung\iot-gateway-opc-ua-proxy\api\csharp\samples\provider\relay\src\Microsoft.Azure.Devices.Proxy.Relay\Microsoft.Azure.Devices.Proxy.Relay.csproj]
C:\Program Files\dotnet\sdk\1.0.4\Microsoft.Common.CurrentVersion.targets(1964,5): warning MSB3277: Found conflicts between different versions of the same dependent assembly that could not be resolved.  These reference conflicts are listed in the build log when log verbosity is set to detailed. [D:\Programmierung\iot-gateway-opc-ua-proxy\api\csharp\samples\opc-ua\test\Client\Client.csproj]
    2 Warning(s)
    0 Error(s)

Time Elapsed 00:00:33.48
122 Datei(en) kopiert

Microsoft (R) Build Engine version 15.1.1012.6693
Copyright (C) Microsoft Corporation. All rights reserved.

  Microsoft.Azure.Devices.Proxy -> D:\Programmierung\iot-gateway-opc-ua-proxy\api\csharp\src\Microsoft.Azure.Devices.Proxy\bin\Debug\net46\Microsoft.Azure.Devices.Proxy.dll
  Microsoft.Azure.Devices.Proxy.Relay -> D:\Programmierung\iot-gateway-opc-ua-proxy\api\csharp\samples\provider\relay\src\Microsoft.Azure.Devices.Proxy.Relay\bin\Debug\net46\Microsoft.Azure.Devices.Proxy.Relay.dll
  Microsoft.Azure.Devices.Proxy -> D:\Programmierung\iot-gateway-opc-ua-proxy\api\csharp\src\Microsoft.Azure.Devices.Proxy\bin\Debug\net45\Microsoft.Azure.Devices.Proxy.dll
  Microsoft.Azure.Devices.Proxy.Relay -> D:\Programmierung\iot-gateway-opc-ua-proxy\api\csharp\samples\provider\relay\src\Microsoft.Azure.Devices.Proxy.Relay\bin\Debug\net45\Microsoft.Azure.Devices.Proxy.Relay.dll
  Microsoft.Azure.Devices.Proxy -> D:\Programmierung\iot-gateway-opc-ua-proxy\api\csharp\src\Microsoft.Azure.Devices.Proxy\bin\Debug\netstandard1.3\Microsoft.Azure.Devices.Proxy.dll
[... cut ...]
  Microsoft.Azure.Devices.Proxy -> D:\Programmierung\iot-gateway-opc-ua-proxy\api\csharp\src\Microsoft.Azure.Devices.Proxy\bin\Debug\netstandard1.3\Microsoft.Azure.Devices.Proxy.dll
  Microsoft.Azure.Devices.Proxy.AspNet -> D:\Programmierung\iot-gateway-opc-ua-proxy\api\csharp\samples\provider\webapp\src\Microsoft.Azure.Devices.Proxy.AspNet\bin\Debug\netstandard1.3\Microsoft.Azure.Devices.Proxy.AspNet.dll

Build succeeded.
    0 Warning(s)
    0 Error(s)

Time Elapsed 00:00:07.85
40 Datei(en) kopiert

```

marcschier, 2017-07-21:
Issue I believe was due to httpclient not being thread safe.  This happens mostly with .net 46 build of opc client, but also with .net standard (but less often).  Making each thread (task) have its own client should address this.  Fixed in #58

JuergenKosel, 2017-07-21:
Hello,

even with #58 , the opc ua sample client crashes by a stack over flow.
There are also still some nuget conflict warnings. Do I needto wait for an update of the nugets, before retesting?

```
  Microsoft.Azure.Devices.Proxy -> D:\Programmierung\iot-gateway-opc-ua-proxy\api\csharp\src\Microsoft.Azure.Devices.Proxy\bin\Debug\netstandard1.3\Microsoft.Azure.Devices.Proxy.dll
C:\Program Files\dotnet\sdk\1.0.4\Microsoft.Common.CurrentVersion.targets(1964,5): warning MSB3277: Found conflicts between different versions of the same dependent assembly that could not be resolved.  These reference conflicts are listed in the build log when log verbosity is set to detailed. [D:\Programmierung\iot-gateway-opc-ua-proxy\api\csharp\samples\provider\relay\src\Microsoft.Azure.Devices.Proxy.Relay\Microsoft.Azure.Devices.Proxy.Relay.csproj]
  Microsoft.Azure.Devices.Proxy.Relay -> D:\Programmierung\iot-gateway-opc-ua-proxy\api\csharp\samples\provider\relay\src\Microsoft.Azure.Devices.Proxy.Relay\bin\Debug\netstandard1.3\Microsoft.Azure.Devices.Proxy.Relay.dll
C:\Program Files\dotnet\sdk\1.0.4\Microsoft.Common.CurrentVersion.targets(1964,5): warning MSB3277: Found conflicts between different versions of the same dependent assembly that could not be resolved.  These reference conflicts are listed in the build log when log verbosity is set to detailed. [D:\Programmierung\iot-gateway-opc-ua-proxy\api\csharp\samples\netcat\Netcat.csproj]
  Netcat -> D:\Programmierung\iot-gateway-opc-ua-proxy\api\csharp\samples\netcat\bin\Debug\netcoreapp1.1\PNetcat.dll
  Microsoft.Azure.Devices.Proxy -> D:\Programmierung\iot-gateway-opc-ua-proxy\api\csharp\src\Microsoft.Azure.Devices.Proxy\bin\Debug\net46\Microsoft.Azure.Devices.Proxy.dll
  Microsoft.Azure.Devices.Proxy.Relay -> D:\Programmierung\iot-gateway-opc-ua-proxy\api\csharp\samples\provider\relay\src\Microsoft.Azure.Devices.Proxy.Relay\bin\Debug\net46\Microsoft.Azure.Devices.Proxy.Relay.dll
  Netcat -> D:\Programmierung\iot-gateway-opc-ua-proxy\api\csharp\samples\netcat\bin\Debug\net46\PNetcat.exe

Build succeeded.

C:\Program Files\dotnet\sdk\1.0.4\Microsoft.Common.CurrentVersion.targets(1964,5): warning MSB3277: Found conflicts between different versions of the same dependent assembly that could not be resolved.  These reference conflicts are listed in the build log when log verbosity is set to detailed. [D:\Programmierung\iot-gateway-opc-ua-proxy\api\csharp\samples\provider\relay\src\Microsoft.Azure.Devices.Proxy.Relay\Microsoft.Azure.Devices.Proxy.Relay.csproj]
C:\Program Files\dotnet\sdk\1.0.4\Microsoft.Common.CurrentVersion.targets(1964,5): warning MSB3277: Found conflicts between different versions of the same dependent assembly that could not be resolved.  These reference conflicts are listed in the build log when log verbosity is set to detailed. [D:\Programmierung\iot-gateway-opc-ua-proxy\api\csharp\samples\netcat\Netcat.csproj]
    2 Warning(s)
    0 Error(s)
```

```
  Successfully created package 'D:\Programmierung\iot-gateway-opc-ua-proxy\build\Debug\Microsoft.Azure.Devices.Proxy.1.0.2.nupkg'.
... Success

D:\Programmierung\iot-gateway-opc-ua-proxy\bld>cd D:\Programmierung\iot-gateway-opc-ua-proxy\api\csharp\samples\opc-ua\test\Client

D:\Programmierung\iot-gateway-opc-ua-proxy\api\csharp\samples\opc-ua\test\Client>dotnet bin\Debug\netcoreapp1.1\Client.dll --url opc.tcp://[fd00:42:2001:db8:206:71ff:fe33:12]:4840/
.Net Core OPC UA Console Client
1 - Create an Application Configuration.
2 - Discover endpoints of opc.tcp://[fd00:42:2001:db8:206:71ff:fe33:12]:4840/.

Process is terminated due to StackOverflowException.
```

marcschier, 2017-07-21:
The warning is not related.  Investigating.

marcschier, 2017-07-21:
So - I was able to reproduce this yesterday and then thought I had narrowed it down to httpclient.  After giving each caller their own httpclient instance I was not able to reproduce this anymore - case closed.  Today I see this issue also - again - but on a different machine, yet even running it from VS.  I deleted the "OPC Foundation" folder, and then it worked.   Investigating more.

marcschier, 2017-07-21:
#59

closed 2017-07-21 by marcschier