Azure/iot-edge-opc-proxy #56: proxyd looses connection and can not reconnect

JuergenKosel, 2017-07-19:
Hello,

while running proxyd over night on my embedded Linux device, I see the problem that proxyd looses the connection to the azure cloud.
And even a new instance of the proxyd program can not reconnect. Only a reboot of the embedded Linux device allows to reconnect. Maybe this is related to the port number assigned by the OS for the outgoing connection.
Here the output from my device syslog:

```
Jul 18 14:11:37 localhost proxyd: === Azure iot-edge-reverse-proxy 1.0.1.1.0.0-7-g5e5fd1d5 ===
Jul 18 14:11:37 localhost proxyd: 
Jul 18 14:11:37 localhost proxyd: Version: 1.0.1.1.0.0-7-g5e5fd1d5
Jul 18 14:11:37 localhost proxyd: Tue Jul 18 14:04:17 2017 (prx_host_init:1029) [DEBUG] Proxy host created!
Jul 18 14:11:37 localhost proxyd: Tue Jul 18 14:04:17 2017 (pal_wsclient_lws_log:122) [INFO] Initial logging level -1
Jul 18 14:11:37 localhost proxyd: 
Jul 18 14:11:37 localhost proxyd: Tue Jul 18 14:04:17 2017 (pal_wsclient_lws_log:122) [INFO] Libwebsockets version: 2.1.0 klj@wklj-v2.0.0-239-g24e77a04
Jul 18 14:11:37 localhost proxyd: 
Jul 18 14:11:37 localhost proxyd: Tue Jul 18 14:04:17 2017 (pal_wsclient_lws_log:122) [INFO] IPV6 compiled in and enabled
Jul 18 14:11:37 localhost proxyd: 
Jul 18 14:11:37 localhost proxyd: Tue Jul 18 14:04:17 2017 (pal_wsclient_lws_log:122) [INFO] libev support not compiled in
Jul 18 14:11:37 localhost proxyd: 
Jul 18 14:11:37 localhost proxyd: Tue Jul 18 14:04:17 2017 (pal_wsclient_lws_log:122) [INFO] libuv support not compiled in
Jul 18 14:11:37 localhost proxyd: 
Jul 18 14:11:37 localhost proxyd: Tue Jul 18 14:04:17 2017 (pal_wsclient_lws_log:122) [INFO]  Threads: 1 each 65536 fds
Jul 18 14:11:37 localhost proxyd: 
Jul 18 14:11:37 localhost proxyd: Tue Jul 18 14:04:17 2017 (pal_wsclient_lws_log:122) [INFO]  mem: platform fd map: 262144 bytes
Jul 18 14:11:37 localhost proxyd: 
Jul 18 14:11:37 localhost proxyd: Tue Jul 18 14:04:17 2017 (pal_wsclient_lws_log:122) [INFO]  Compiled with OpenSSL support
Jul 18 14:11:37 localhost proxyd: 
Jul 18 14:11:37 localhost proxyd: Tue Jul 18 14:04:17 2017 (pal_wsclient_lws_log:122) [INFO] Doing SSL library init
Jul 18 14:11:37 localhost proxyd: 
Jul 18 14:11:37 localhost proxyd: Tue Jul 18 14:04:17 2017 (pal_wsclient_lws_log:122) [INFO] Creating Vhost 'default' port -1, 1 protocols, IPv6 on
Jul 18 14:11:37 localhost proxyd: 
Jul 18 14:11:37 localhost proxyd: Tue Jul 18 14:04:17 2017 (pal_wsclient_lws_log:122) [INFO]  mem: per-conn:          448 bytes + protocol rx buf
Jul 18 14:11:37 localhost proxyd: 
Jul 18 14:11:37 localhost proxyd: Tue Jul 18 14:04:17 2017 (pal_wsclient_lws_log:122) [INFO]  canonical_hostname = uagate0012
Jul 18 14:11:37 localhost proxyd: 
Jul 18 14:11:37 localhost proxyd: Tue Jul 18 14:04:17 2017 (pal_sdclient_callback:335) [DEBUG] Avahi connected, server running!
Jul 18 14:11:37 localhost proxyd: Tue Jul 18 14:04:17 2017 (prx_module_create:117) [DEBUG] Proxy module created in GW Host!
Jul 18 14:11:37 localhost proxyd: Tue Jul 18 14:04:17 2017 (prx_host_start:807) [INFO] Host server started!
Jul 18 14:11:37 localhost proxyd: Tue Jul 18 14:04:17 2017 (pal_wsclient_lws_log:122) [INFO] lws_ssl_client_connect2: SSL_connect says -1
Jul 18 14:11:37 localhost proxyd: 
Jul 18 14:11:37 localhost proxyd: Tue Jul 18 14:04:17 2017 (pal_wsclient_lws_log:122) [INFO] lws_ssl_client_connect2: SSL_connect says -1
Jul 18 14:11:37 localhost proxyd: 
Jul 18 14:11:37 localhost proxyd: Tue Jul 18 14:04:17 2017 (pal_wsclient_lws_log:122) [INFO] lws_ssl_client_connect2: SSL_connect says 1
Jul 18 14:11:37 localhost proxyd: 
Jul 18 14:11:37 localhost proxyd: Tue Jul 18 14:04:18 2017 (io_mqtt_connection_handle_CONNECT_ACK:735) [INFO] Connection established (softingconnectedfactoryf1605.azure-devices.net:443)!
Jul 18 14:11:37 localhost proxyd: Tue Jul 18 14:04:18 2017 (io_mqtt_connection_handle_SUBSCRIBE_ACK:677) [DEBUG] Subscribed to $iothub/methods/POST/#!
Jul 18 14:11:37 localhost proxyd: Tue Jul 18 14:04:37 2017 (io_mqtt_connection_operation_callback:824) [INFO] Connection alive...
Jul 18 14:11:37 localhost proxyd: Tue Jul 18 14:04:57 2017 (io_mqtt_connection_operation_callback:824) [INFO] Connection alive...
```
cut

```
l 18 15:16:37 localhost proxyd: Tue Jul 18 15:15:47 2017 (io_mqtt_connection_operation_callback:824) [INFO] Connection alive...
Jul 18 15:16:37 localhost proxyd: Tue Jul 18 15:16:17 2017 (io_mqtt_connection_operation_callback:824) [INFO] Connection alive...
Jul 18 15:17:01 localhost CRON[16609]: (root) CMD (   cd / && run-parts --report /etc/cron.hourly)
Jul 18 15:21:31 localhost avahi-daemon[1497]: Leaving mDNS multicast group on interface eth0.IPv6 with address 2002:3e92:37e2:0:206:71ff:fe33:12.
Jul 18 15:21:31 localhost avahi-daemon[1497]: Joining mDNS multicast group on interface eth0.IPv6 with address fd00:42:2001:db8:206:71ff:fe33:12.
Jul 18 15:21:32 localhost ntpd[1479]: Deleting interface #5 eth0, 2002:3e92:37e2:0:206:71ff:fe33:12#123, interface stats: received=0, sent=0, dropped=0, active_time=4669 secs
Jul 18 15:21:32 localhost ntpd[1479]: peers refreshed
Jul 18 15:23:59 localhost proxyd: Tue Jul 18 15:16:37 2017 (io_mqtt_connection_operation_callback:824) [INFO] Connection alive...
Jul 18 15:23:59 localhost proxyd: Tue Jul 18 15:16:57 2017 (io_mqtt_connection_operation_callback:824) [INFO] Connection alive...
Jul 18 15:23:59 localhost proxyd: Tue Jul 18 15:17:27 2017 (io_mqtt_connection_operation_callback:824) [INFO] Connection alive...
Jul 18 15:23:59 localhost proxyd: Tue Jul 18 15:17:57 2017 (io_mqtt_connection_operation_callback:824) [INFO] Connection alive...
Jul 18 15:23:59 localhost proxyd: Tue Jul 18 15:18:17 2017 (io_mqtt_connection_operation_callback:824) [INFO] Connection alive...
Jul 18 15:23:59 localhost proxyd: Tue Jul 18 15:18:47 2017 (io_mqtt_connection_operation_callback:824) [INFO] Connection alive...
Jul 18 15:23:59 localhost proxyd: Tue Jul 18 15:19:17 2017 (io_mqtt_connection_operation_callback:824) [INFO] Connection alive...
Jul 18 15:23:59 localhost proxyd: Tue Jul 18 15:19:37 2017 (io_mqtt_connection_operation_callback:824) [INFO] Connection alive...
Jul 18 15:23:59 localhost proxyd: Tue Jul 18 15:19:57 2017 (io_mqtt_connection_operation_callback:824) [INFO] Connection alive...
Jul 18 15:23:59 localhost proxyd: Tue Jul 18 15:20:27 2017 (io_mqtt_connection_operation_callback:824) [INFO] Connection alive...
Jul 18 15:23:59 localhost proxyd: Tue Jul 18 15:20:47 2017 (io_mqtt_connection_operation_callback:824) [INFO] Connection alive...
[... 258 lines cut ...]
Jul 19 06:20:57 localhost proxyd: 
Jul 19 06:20:57 localhost proxyd: Wed Jul 19 06:18:50 2017 (pal_wsclient_lws_log:122) [INFO]  mem: platform fd map: 262144 bytes
Jul 19 06:20:57 localhost proxyd: 
Jul 19 06:20:57 localhost proxyd: Wed Jul 19 06:18:50 2017 (pal_wsclient_lws_log:122) [INFO]  Compiled with OpenSSL support
Jul 19 06:20:57 localhost proxyd: 
Jul 19 06:20:57 localhost proxyd: Wed Jul 19 06:18:50 2017 (pal_wsclient_lws_log:122) [INFO] Doing SSL library init
Jul 19 06:20:57 localhost proxyd: 
Jul 19 06:20:57 localhost proxyd: Wed Jul 19 06:18:50 2017 (pal_wsclient_lws_log:122) [INFO] Creating Vhost 'default' port -1, 1 protocols, IPv6 on
Jul 19 06:20:57 localhost proxyd: 
Jul 19 06:20:57 localhost proxyd: Wed Jul 19 06:18:50 2017 (pal_wsclient_lws_log:122) [INFO]  mem: per-conn:          448 bytes + protocol rx buf
Jul 19 06:20:57 localhost proxyd: 
Jul 19 06:20:57 localhost proxyd: Wed Jul 19 06:18:50 2017 (pal_wsclient_lws_log:122) [INFO]  canonical_hostname = uagate0012
Jul 19 06:20:57 localhost proxyd: 
Jul 19 06:20:57 localhost proxyd: Wed Jul 19 06:19:20 2017 (io_mqtt_connection_monitor:483) [ERROR] missing pub ack, hard reset...
Jul 19 06:20:57 localhost proxyd: Wed Jul 19 06:19:20 2017 (io_mqtt_connection_complete_disconnect:931) [DEBUG] Socket disconnected.
Jul 19 06:20:57 localhost proxyd: Wed Jul 19 06:19:20 2017 (io_mqtt_connection_complete_disconnect:939) [DEBUG] Client disconnected.
Jul 19 06:20:57 localhost proxyd: Wed Jul 19 06:19:20 2017 (prx_server_handler:2415) [DEBUG] Server ctrl is reconnecting in 16 sec.(er_writing)
Jul 19 06:20:57 localhost proxyd: Wed Jul 19 06:19:20 2017 (io_mqtt_connection_hard_reset:1135) [INFO] Reconnecting in 16 seconds...
Jul 19 06:20:57 localhost proxyd: Wed Jul 19 06:20:06 2017 (io_mqtt_connection_monitor:483) [ERROR] missing pub ack, hard reset...
Jul 19 06:20:57 localhost proxyd: Wed Jul 19 06:20:06 2017 (io_mqtt_connection_complete_disconnect:931) [DEBUG] Socket disconnected.
Jul 19 06:20:57 localhost proxyd: Wed Jul 19 06:20:06 2017 (io_mqtt_connection_complete_disconnect:939) [DEBUG] Client disconnected.
Jul 19 06:20:57 localhost proxyd: Wed Jul 19 06:20:06 2017 (prx_server_handler:2415) [DEBUG] Server ctrl is reconnecting in 32 sec.(er_writing)
Jul 19 06:20:57 localhost proxyd: Wed Jul 19 06:20:06 2017 (io_mqtt_connection_hard_reset:1135) [INFO] Reconnecting in 32 seconds...

```

JuergenKosel, 2017-07-19:
This problem is caused by local network misconfiguration -> closing

closed 2017-07-19 by JuergenKosel

merlingeorge555, 2018-09-26:
Please help to solve this issue @JuergenKosel . What is the local network misconfiguration?

[Pid=1:Tid=0x7a502ae8 14:45:20.317] io_mqtt ERROR Connection error, reset connection... [io_mqtt_connection_error_callback:860]
[Pid=1:Tid=0x7a502ae8 14:45:20.418] xlog NOTICE Closing tlsio from a state other than TLSIO_STATE_EXT_OPEN or TLSIO_STATE_EXT_ERROR [tlsio_openssl_close:1231]
[Pid=1:Tid=0x7a502ae8 14:45:20.418] xlog NOTICE Closing tlsio from a state other than TLSIO_STATE_EXT_OPEN or TLSIO_STATE_EXT_ERROR [tlsio_openssl_close:1231]
[Pid=1:Tid=0x7a502ae8 14:45:20.425] io_mqtt INFO Socket disconnected. [io_mqtt_connection_complete_disconnect:931]
[Pid=1:Tid=0x7a502ae8 14:45:20.425] io_mqtt INFO Client disconnected. [io_mqtt_connection_complete_disconnect:939]
[Pid=1:Tid=0x7a502ae8 14:45:20.425] io_mqtt NOTICE Reconnecting... [io_mqtt_connection_hard_reset:1166]
[Pid=1:Tid=0x7a4e7ae8 14:45:20.426] pal_ws INFO Connecting 0x560e00c227e0 to aaa.azure-devices.net:443 (ssl) ... [pal_wsworker_thread:961]
[Pid=1:Tid=0x7a502ae8 14:45:20.659] io_mqtt_sub INFO Subscribed to $iothub/methods/POST/#! [io_mqtt_connection_handle_SUBSCRIBE_ACK:677]
[Pid=1:Tid=0x7a4e7ae8 14:45:20.991] pal_ws_lws INFO lws_ssl_client_connect2: SSL_connect says -1 [pal_wsclient_lws_log:126]
[Pid=1:Tid=0x7a4e7ae8 14:45:20.992] pal_ws_lws INFO lws_ssl_client_connect2: SSL_connect says -1 [pal_wsclient_lws_log:126]
[Pid=1:Tid=0x7a4e7ae8 14:45:21.019] pal_ws_lws INFO lws_ssl_client_connect2: SSL_connect says -1 [pal_wsclient_lws_log:126]
[Pid=1:Tid=0x7a4e7ae8 14:45:21.247] pal_ws_lws INFO lws_ssl_client_connect2: SSL_connect says 1 [pal_wsclient_lws_log:126]
[Pid=1:Tid=0x7a502ae8 14:45:21.968] io_mqtt NOTICE Connection established (aaa.azure-devices.net:443)! [io_mqtt_connection_handle_CONNECT_ACK:735]
[Pid=1:Tid=0x7a502ae8 14:45:22.216] io_mqtt_sub INFO Subscribed to $iothub/methods/POST/#! [io_mqtt_connection_handle_SUBSCRIBE_ACK:677]
[Pid=1:Tid=0x7a4e7ae8 14:45:27.197] pal_ws_lws ERROR lws_ssl_capable_read failed: error:00000005:lib(0):func(0):DH lib [pal_wsclient_lws_log:116]
[Pid=1:Tid=0x7a4e7ae8 14:45:27.198] pal_ws NOTICE lws reported remote side closed [pal_wsworker_lws_on_disconnect:773]
[Pid=1:Tid=0x7a502ae8 14:45:27.199] io_mqtt ERROR Connection error, reset connection... [io_mqtt_connection_error_callback:860]
[Pid=1:Tid=0x7a502ae8 14:45:27.300] io_mqtt INFO Socket disconnected. [io_mqtt_connection_complete_disconnect:931]
[Pid=1:Tid=0x7a502ae8 14:45:27.300] io_mqtt INFO Client disconnected. [io_mqtt_connection_complete_disconnect:939]
[Pid=1:Tid=0x7a502ae8 14:45:27.300] io_mqtt NOTICE Reconnecting... [io_mqtt_connection_hard_reset:1166]
[Pid=1:Tid=0x7a502ae8 14:45:27.333] pal_sk NOTICE Resolving aaa699aa.azure-devices.net:8883... [pal_socket_open_by_name:1016]
[Pid=1:Tid=0x7a502ae8 14:45:27.333] root NOTICE Resolving aaa699aa.azure-devices.net:8883 (family: 0)... [pal_getaddrinfo:1025]
[Pid=1:Tid=0x7a502ae8 14:45:27.394] root INFO   ... aaa699aa.azure-devices.net:8883 -> 104.40.49.44:8883 (ihsu-prod-by-002.cloudapp.net) [pal_net_trace_resolved:35]
[Pid=1:Tid=0x7a502ae8 14:45:27.394] root INFO   ... aaa.azure-devices.net:8883 -> 104.40.49.44:8883 (ihsu-prod-by-002.cloudapp.net) [pal_net_trace_resolved:35]
[Pid=1:Tid=0x7a502ae8 14:45:27.394] pal_sk INFO Socket created ... (fd:10) [pal_socket_try_open:865]
[Pid=1:Tid=0x7a502ae8 14:45:27.394] pal_sk INFO Socket connecting ... (fd:10) [pal_socket_connect:723]
[Pid=1:Tid=0x7a502ae8 14:45:27.618] pal_sk INFO Socket connected synchronously! [pal_socket_connect:753]
[Pid=1:Tid=0x7a502ae8 14:45:27.618] pal_sk INFO Socket 10 opened synchronously! [pal_socket_try_next:964]
[Pid=1:Tid=0x7a502ae8 14:45:28.330] io_mqtt NOTICE Connection established (aaa.azure-devices.net:8883)! [io_mqtt_connection_handle_CONNECT_ACK:735]

merlingeorge555, 2018-09-26:
I resolved MqTT connection issues Now facing this