Azure/iot-edge-opc-proxy #63: OPC UA sample client crashes, if server does not trust client certificate

JuergenKosel, 2017-08-04:
Hello,

OPC UA sample client crashes, if server does not trust client certificate:

```
dotnet bin\Debug\netcoreapp1.1\Client.dll --url "opc.tcp://192.168.25.156:5555"
.Net Core OPC UA Console Client
1 - Create an Application Configuration.
2 - Discover endpoints of opc.tcp://192.168.25.156:5555.
    Selected endpoint uses: Basic256
3 - Create session with OPC UA server.
Accepted Certificate: E=support.automation@softing.com, CN=OPC_UA_Server, OU=IAD, O=Softing, L=Nuernberg, S=Bayern, C=DE
Exit due to Exception: System.AggregateException: One or more errors occurred. (Error establishing a connection: Error received from remote host: ) ---> Opc.Ua.ServiceResultException: Error establishing a connection: Error received from remote host:
   at Opc.Ua.Client.Session.<Create>d__72.MoveNext() in d:\a\1\s\SampleApplications\SDK\Opc.Ua.Client\Session.cs:line 810
--- End of stack trace from previous location where exception was thrown ---
   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw()
   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)
   at Opc.Ua.Client.Session.<Create>d__71.MoveNext() in d:\a\1\s\SampleApplications\SDK\Opc.Ua.Client\Session.cs:line 719
--- End of stack trace from previous location where exception was thrown ---
   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw()
   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)
   at System.Runtime.CompilerServices.TaskAwaiter`1.GetResult()
   at NetCoreConsoleClient.Program.<CreateSessionAsync>d__4.MoveNext() in D:\Programmierung\iot-gateway-opc-ua-proxy\api\csharp\samples\opc-ua\test\Client\Program.cs:line 326
--- End of stack trace from previous location where exception was thrown ---
   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw()
   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)
   at System.Runtime.CompilerServices.TaskAwaiter`1.GetResult()
   at NetCoreConsoleClient.Program.<ConsoleSampleClient>d__3.MoveNext() in D:\Programmierung\iot-gateway-opc-ua-proxy\api\csharp\samples\opc-ua\test\Client\Program.cs:line 237
   --- End of inner exception stack trace ---
   at System.Threading.Tasks.Task.ThrowIfExceptional(Boolean includeTaskCanceledExceptions)
   at System.Threading.Tasks.Task.Wait(Int32 millisecondsTimeout, CancellationToken cancellationToken)
   at System.Threading.Tasks.Task.Wait()
   at NetCoreConsoleClient.Program.Main(String[] args) in D:\Programmierung\iot-gateway-opc-ua-proxy\api\csharp\samples\opc-ua\test\Client\Program.cs:line 149
---> (Inner Exception #0) Opc.Ua.ServiceResultException: Error establishing a connection: Error received from remote host:
   at Opc.Ua.Client.Session.<Create>d__72.MoveNext() in d:\a\1\s\SampleApplications\SDK\Opc.Ua.Client\Session.cs:line 810
--- End of stack trace from previous location where exception was thrown ---
   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw()
   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)
   at Opc.Ua.Client.Session.<Create>d__71.MoveNext() in d:\a\1\s\SampleApplications\SDK\Opc.Ua.Client\Session.cs:line 719
--- End of stack trace from previous location where exception was thrown ---
   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw()
   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)
   at System.Runtime.CompilerServices.TaskAwaiter`1.GetResult()
   at NetCoreConsoleClient.Program.<CreateSessionAsync>d__4.MoveNext() in D:\Programmierung\iot-gateway-opc-ua-proxy\api\csharp\samples\opc-ua\test\Client\Program.cs:line 326
--- End of stack trace from previous location where exception was thrown ---
   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw()
   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)
   at System.Runtime.CompilerServices.TaskAwaiter`1.GetResult()
   at NetCoreConsoleClient.Program.<ConsoleSampleClient>d__3.MoveNext() in D:\Programmierung\iot-gateway-opc-ua-proxy\api\csharp\samples\opc-ua\test\Client\Program.cs:line 237<---
```

If the client certificate is moved into the trusted folder of the server, the client can connect (without crash).

marcschier, 2017-08-04:
The Service_fault information is thrown as C# Exceptions throughout the stack, and this test client sample simply catches it and prints it for informational purposes, waits for the user to press a key and then exits.  So it hasn't crashed even though the trace above looks "bad".   It might be a better cli experience to just print the exception messages, and omit the stack traces.

marcschier, 2017-08-05:
Changed behavior in 1.0.3 to only print error, not stack traces.

closed 2017-08-05 by marcschier