Azure/iot-edge-opc-proxy #68: Difference between OCP UA Proxy Server and OPC UA Aggregating Server

JayeshThamke, 2017-08-07:
I have two queries:
1. By reading the description of OPC UA Proxy Server, what is the difference between OPC UA proxy server and OPC UA Aggregating Server  `(ServerType = ClientAndServer)` ? Is there any specific documentation that describe more on OPC UA Proxy Server or high level conceptual architecture? Thanks 
2. How can I use/host OPC UA Client in Azure cloud? Is the OPC UA Client Service already available in Microsoft Azure cloud? Please refer the following image attached.

Thank you for your reply,
Jayesh

[image]

marcschier, 2017-08-07:
Hi Jayesh, 

1. Aggregation server and proxy are orthogonal to each other.  The reverse proxy is a means to connect to OPC servers in a local factory network from the cloud using secure channel.  An aggregation server could use this feature to aggregate multiple server address spaces and make them accessible in the cloud as one.

2. Take a look at the opcua client sample contained in the API/csharp/samples folder.  You could run this client in the cloud.  An example of how this can be used in practice you find in our connected factory solution, it's source is also on GitHub.  In the connected factory the browser connects to a local server in the simulated factory and allows you to browse the servers address space.

JayeshThamke, 2017-08-07:
Hello Marc - @marcschier ,

Thank you very much for reply and specifically for briefly explaining concept of OPC UA Reverse Proxy Server.
If I understood it correctly,  **OPC UA Proxy Server** meant to connect on-premises OPC UA Server to cloud hosted OPC UA Client Application via secured connection. **OPC UA Reverse Proxy Server** is to connect from cloud hosted OPC UA Client to on-premises OPC UA Servers. 
So in-short is it how the user sees the proxy app?

In next couple of days I will try out Proxy Server example and Connected Factory example. Till that please do not close this issue.
best regards,
Jayesh

marcschier, 2018-01-31:
Closing - covering documentation issues separately.

closed 2018-01-31 by marcschier