Azure/iot-edge-opc-proxy #69: Docker Build Error and Build on Developer Machine Error on Win7 64bit OS

JayeshThamke, 2017-08-09:
Hello,

Building OPC UA Proxy Server Module using Docker container and github repository causes 

> Incomplete Configurations and errors.

Attachment -> docker built from github error.txt

Also, I have cloned the repository and tried to build using the recommendations for Windows 7 machine. 
That is 
- Installed OpenSSL v0.9.8r 
- Ran 
`C:\github_repos\opcproxy\iot-edge-opc-proxy\bld\build.cmd --use-libwebsockets` 
Then it is strange that, it returned an error for unknown C and CXX identification.

Attachment -> build proxy on windows 7 machine.txt
I have checked that in VS 2015, VC++ module is already installed.

I am pretty new to CMake and Docker. Please correct me if I have missed any step during building the application.

Thank you very much for reply,
Jayesh

[build proxy on windows 7 machine.txt](https://github.com/Azure/iot-edge-opc-proxy/files/1211957/build.proxy.on.windows.7.machine.txt)
[docker built from github error.txt](https://github.com/Azure/iot-edge-opc-proxy/files/1211958/docker.built.from.github.error.txt)
[image]

marcschier, 2017-08-09:
Re Win7, Please try using a vs build environment console window and see if this helps cmakegiven find the compiler in the path.  Also, I advise to use OpenSSl 1.0.2k or l, 0.9.8 is pretty old.  If our docs mention this it must be changed.  

Re docker. Normally docker clones all submodules when run the way you did, so this is surprising, also given that dockerhub is building just fine.  I suggest first cloning --recursive, and then running docker build -t proxyd . in the root rather than directly against the git url.  

Let me know if you are still blocked.

JayeshThamke, 2017-08-10:
1. Win 7 + using Docker Container:
I have followed your recommended solution for Docker building, that is changing working directory from Docker Terminal and did `docker build -t proxyd .` it built and ran successfully. Thank you very much. 

2. On my Windows 7 machine, VS2015 have installed and it uses `PlatformToolSet = v140` however in CMakeError.Log inside 'C:\github_repos\opcproxy\iot-edge-opc-proxy\build\cmake\x64\CMakeFiles'
The error was logged as: 

> Build started 10.08.2017 15:11:05.
Project "C:\github_repos\opcproxy\iot-edge-opc-proxy\build\cmake\x64\CMakeFiles\3.8.1\CompilerIdC\CompilerIdC.vcxproj" on node 1 (default targets).
C:\Program Files (x86)\MSBuild\Microsoft.Cpp\v4.0\V140\Microsoft.Cpp.Platform.targets(57,5): error MSB8020: The build tools for v141 (Platform Toolset = 'v141') cannot be found. To build using the v141 build tools, please install v141 build tools.  Alternatively, you may upgrade to the current Visual Studio tools by selecting the Project menu or right-click the solution, and then selecting "Retarget solution". [C:\github_repos\opcproxy\iot-edge-opc-proxy\build\cmake\x64\CMakeFiles\3.8.1\CompilerIdC\CompilerIdC.vcxproj]
Done Building Project "C:\github_repos\opcproxy\iot-edge-opc-proxy\build\cmake\x64\CMakeFiles\3.8.1\CompilerIdC\CompilerIdC.vcxproj" (default targets) -- FAILED.

Error log attached: 
[CMakeError.txt](https://github.com/Azure/iot-edge-opc-proxy/files/1215502/CMakeError.txt)

Is the C/C++ Compiler version depends on 'PlatformToolSet' version? Did I receive above error because of it? I am sorry to say but I cannot upgrade VS2015 to VS2017.  Thanks for inputs,

3. From fig. 1 below, Proxy module communicates with IoTHub using AMQP telemetry (possibly via port 5671 over TLS), 
       3.a. however running sample example connects with MQTT via port 8883 over SSL why so? Please 
       note that I have not yet connected (On-Premises) OPC UA Server to the proxy. 

[image]

3.b. While the Proxy Module is running 

> Connection alive message was published to IoTHub, as follows:

> [Pid=1:Tid=0x95bf9ae8 07:29:03.739] io_mqtt NOTICE Connection alive... [io_mqtt_
connection_operation_callback:824]
[Pid=1:Tid=0x95bf9ae8 07:29:23.743] io_mqtt NOTICE Connection alive... [io_mqtt_
connection_operation_callback:824]
[Pid=1:Tid=0x95bf9ae8 07:29:33.926] io_mqtt NOTICE Connection alive... [io_mqtt_
connection_operation_callback:824]

Is it the Proxy Server element continuously publishes these messages?

4. I have observed that the Proxy Module (a device) created in IoTHub but its "Device Id" looks like a pointer handle value e.g. 661a630988b3 . I would like to create the Proxy Module (a device) with some meaningful name, is it possible? Please tell me where and how the changes need to be done. Thank you.

5. Where can I find the log repository generated from zlog module?

marcschier, 2017-08-11:
Hi @JayeshThamke, 

1. Excellent
2. VS 2015 should work just fine.  cmake 3.8 seems to default now to VS 2017 toolchain, but you can use the -t option of build.cmd to specify v140.  Regarding cmake not finding the compiler, this looks to me to be a PATH/config issue.  There are multiple VS 2015 prompts, please use the x64 or x86 native command prompt, not the general one and see if that helps.
3. We changed from AMQP to MQTT when we switched to Device methods some time last year.  We, Azure, just release AMQP support for Device methods, so AMQP would be a possibility now as well. If you have business reasons for AMQP to be used in the proxy, please open a feature request.  If you are asking about websocket support, the proxy will fall back to websocket when 8883 is not accessible.  You can force using only websocket using the -w command line option.
3b. These logged as a result of a MQTT PING ACK, so only MQTT protocol related and do not account for "messages" to IoTHub.  It shows you that the mqtt connection is running.  Without ACK mqtt will reconnect.
4. the proxy is installed under the name of the host it is running on (gethostname).  If you run in docker, docker assigns a random id as host name, which you see here.  It does so every time you start, so you will accrue proxies.  It is thus advisable to either a) run the container using -h option and assign a name, or b) use the --name option when installing the proxy in IoT hub using -I.
5. You mean the log file?  You can specify a log file name on the command line (see --help), when compiling in zlog.  Otherwise, if you are asking about the git repo, it is included as submodule (see .gitmodules).

JayeshThamke, 2017-08-11:
Hello @marcschier ,
RE 2. Thanks for suggestion, I built the Proxy module using VS2015 x64 Native Tools Command Prompt with -T v140 , and this time C and C++ compilers are identified. However, I still get some errors regarding linking binaries 
> Unresolved External Symbol
Please see the attachment for details - (I have --skipped-unittests)
[unresolved external symbol.txt](https://github.com/Azure/iot-edge-opc-proxy/files/1218018/unresolved.external.symbol.txt)

I have tried to manually add 'NCrypto.lib' in Linker->Input of the project where the error has occurred but did not work.

RE 3.
3a. Thanks for detail and no I don't habe any business condition for AMQP, MQTT will work for me. Could you please tell me how can I run '-w' command (the complete command line) & where can I find the description? 
3b. Thanks!

RE 4. Thanks, I got it!
RE 5. Yes I want to see the log file name. In help do you mean --use-zlog command for setting up zlogger? Is the name or path configurable for this command? Thanks
Attachment below:
[image]

-Jayesh

marcschier, 2017-08-11:
2. That is a bug.  Ncrypt should not be involved here.  We need to look into this, likely an issue in how iotsdk is built, i.e. it does not honor the openssl flags.  
3a. "proxyd.exe -w"  Note that you need to specify the connection string, which is in the samples done through ENV VAR _HUB_CS.
5. Yes, --use-zlog will compile zlog in (default if you open proxyd.sln, but that will not work on Win7).   It is included as submodule, so no more needed.  If you do not build with it, it will use xlogging, i.e. just console logging (which might be ok as well if you scrape the stdout).  If you compile zlog in, then you can run "proxyd.exe -l <logfilename>" to write to log file (in addition to stdout).  Use the -L option to specify a zlog config file (see zlog help), which allows you to control all logging, including logging to syslog on Linux, not logging to stdout, log formats, etc..  However, rotation does not work on Windows.

marcschier, 2017-08-12:
Changed title to reflect 2.

JayeshThamke, 2017-08-14:
Hello @marcschier 
Thank you very much for detailed analysis, I would like to know if you have any timeline to fix the issue.
Thanks,
Jayesh

marcschier, 2017-08-14:
#71

marcschier, 2017-08-14:
Hi @JayeshThamke, 

I am going to fix the build error and reenable the --use-openssl and --use-lws gates on appveyor (see #71).  Since appveyor uses a 2012 server image to build, can you see if this works for you on Win7?

Thx.

Marc

closed 2017-08-14 by marcschier

JayeshThamke, 2017-08-14:
Hello Marc @marcschier 
On Old git cloned - I can now build the "proxy.sln" solution  by **manually opening it in VS2015 and changing its** 

> Target Platform Version = 8.1 (it is default settings when I create any new VC++ project in Win7 and VS2015)

 and 

> Platform Toolset = v140 

from the projects' properties.

Rebuilding the project does not give any error but running by simply hitting F5 pops up 

> Entry Point Not Found error window (please see attached)

[image]

Winhttp.dll is available for Minimum Win 8.1

marcschier, 2017-08-14:
Yes, this is by design and hence that is why --use-libwebsockets will be needed on Win7.  Websocket support was first added to winhttp in Win8.   However, you can also work around this by replacing pal_ws_win.c with pal_ws_stub.c but then would not have websocket transport support for mqtt (only port 8883).  If libwebsockets doesn't work for you, pal_ws.h can be implemented wrapping other websocket libraries, e.g. uws_client.c in azure-c-shared-utility, contributions would be welcome.

asgaralim, 2019-03-05:
I configured the system as said the github.  I have opc ua server, opc proxy ( configured as iot edge module and run as docker container) and opc ua client on same machine. I am unable to establish connected with opc ua server. Could you please provide detailed document to configure above components.

JayeshThamke, 2019-03-05:
@asgaralim Is there problem with identification of peers? Once the certificates are generated you need to manually copy the public keys into certificate store.