Azure/iot-edge-opc-proxy #73: How to run the sample in separate machines?

606simps, 2017-08-16:
Hi,
I tried the proxy module with this [sample](https://github.com/Azure/iot-edge-opc-proxy/blob/master/api/csharp/samples/opc-ua/readme.md) all running on the same machine, and everything works fine.
Now I am trying to set up the sample client on a separate machine from the proxy and Opc Ua server.
I have sample client running on machine 1. Opc Ua server and proxy module running on machine 2.
Machine 1 is connected to the internet directly, machine 2 is connected to a wireless AP which connects to the internet. I am getting these error logs...

Client
.Net Core OPC UA Console Client
1 - Create an Application Configuration.
2 - Discover endpoints of opc.tcp://192.168.1.100:51210/UA/SampleServer.
    Selected endpoint uses: Basic256Sha256
3 - Create session with OPC UA server.
One or more error(s) occurred:
   -> Error establishing a connection: BadNotConnected
Press any key to exit

Proxy
[Pid=1:Tid=0xd9347ae8 03:24:52.710] root NOTICE Resolving 192.168.1.100:51210 (family: 0)... [pal_getaddrinfo:1180]
[Pid=1:Tid=0xd9347ae8 03:24:52.710] root INFO   ... 192.168.1.100:51210 -> 192.168.1.100:51210 (192.168.1.100) [pal_net_trace_resolved:36]
[Pid=1:Tid=0xd9347ae8 03:24:52.710] root INFO   ... 192.168.1.100:51210 -> 192.168.1.100:51210 (192.168.1.100) [pal_net_trace_resolved:36]
[Pid=1:Tid=0xd9347ae8 03:24:52.711] tp_mqtt NOTICE OUT: [PING#1] [io_iot_hub_umqtt_connection_on_send:329]
[Pid=1:Tid=0xd9347ae8 03:24:53.288] tp_mqtt NOTICE IN: [LINK#2] [io_iot_hub_umqtt_connection_on_receive:451]
[Pid=1:Tid=0xd9347ae8 03:24:53.288] pal_sk NOTICE Resolving 192.168.1.100:51210... [pal_socket_open_by_name:1016]
[Pid=1:Tid=0xd9347ae8 03:24:53.288] root NOTICE Resolving 192.168.1.100:51210 (family: 2)... [pal_getaddrinfo:1180]
[Pid=1:Tid=0xd9347ae8 03:24:53.288] root INFO   ... 192.168.1.100:51210 -> 192.168.1.100:51210 (192.168.1.100) [pal_net_trace_resolved:36]
[Pid=1:Tid=0xd9347ae8 03:24:53.288] root INFO   ... 192.168.1.100:51210 -> 192.168.1.100:51210 (192.168.1.100) [pal_net_trace_resolved:36]
[Pid=1:Tid=0xd9347ae8 03:24:53.288] pal_sk INFO Socket created ... (fd:16) [pal_socket_try_open:865]
[Pid=1:Tid=0xd9347ae8 03:24:53.288] pal_sk INFO Socket connecting ... (fd:16) [pal_socket_connect:723]
[Pid=1:Tid=0xd9347ae8 03:24:53.305] pal_sk INFO Socket connected synchronously! [pal_socket_connect:753]
[Pid=1:Tid=0xd9347ae8 03:24:53.305] pal_sk INFO Socket 16 opened synchronously! [pal_socket_try_next:964]
[Pid=1:Tid=0xd9347ae8 03:24:53.306] tp_mqtt NOTICE OUT: [LINK#2] [io_iot_hub_umqtt_connection_on_send:329]
[Pid=1:Tid=0xd9347ae8 03:24:53.824] tp_mqtt NOTICE IN: [OPEN#3] [io_iot_hub_umqtt_connection_on_receive:451]
[Pid=1:Tid=0xd9347ae8 03:24:53.824] server_sk NOTICE Socket open! [prx_server_socket_handle_openrequest:1509]
[Pid=1:Tid=0xd9347ae8 03:24:53.824] tp_mqtt NOTICE OUT: [OPEN#3] [io_iot_hub_umqtt_connection_on_send:329]
[Pid=1:Tid=0xd9347ae8 03:24:54.359] tp_mqtt INFO IN: [POLL#4] [io_iot_hub_umqtt_connection_on_receive:446]
[Pid=1:Tid=0xd9347ae8 03:24:54.483] tp_mqtt INFO IN: [DATA#5] [io_iot_hub_umqtt_connection_on_receive:446]
[Pid=1:Tid=0xd9347ae8 03:24:54.485] tp_mqtt INFO OUT: [POLL#5] [io_iot_hub_umqtt_connection_on_send:324]
[Pid=1:Tid=0xd9347ae8 03:24:54.524] tp_mqtt INFO OUT: [DATA#0] [io_iot_hub_umqtt_connection_on_send:324]
[Pid=1:Tid=0xd9347ae8 03:24:54.779] tp_mqtt INFO IN: [POLL#6] [io_iot_hub_umqtt_connection_on_receive:446]
[Pid=1:Tid=0xd9347ae8 03:24:54.825] tp_mqtt INFO IN: [DATA#7] [io_iot_hub_umqtt_connection_on_receive:446]
[Pid=1:Tid=0xd9347ae8 03:24:54.827] tp_mqtt INFO OUT: [POLL#7] [io_iot_hub_umqtt_connection_on_send:324]
[Pid=1:Tid=0xd9347ae8 03:24:55.028] tp_mqtt INFO OUT: [DATA#2] [io_iot_hub_umqtt_connection_on_send:324]
[Pid=1:Tid=0xd9347ae8 03:24:55.244] tp_mqtt INFO IN: [POLL#8] [io_iot_hub_umqtt_connection_on_receive:446]
[Pid=1:Tid=0xd9347ae8 03:24:55.296] tp_mqtt INFO IN: [DATA#9] [io_iot_hub_umqtt_connection_on_receive:446]
[Pid=1:Tid=0xd9347ae8 03:24:55.298] tp_mqtt INFO OUT: [POLL#9] [io_iot_hub_umqtt_connection_on_send:324]
[Pid=1:Tid=0xd9347ae8 03:24:55.447] tp_mqtt INFO OUT: [DATA#4] [io_iot_hub_umqtt_connection_on_send:324]
[Pid=1:Tid=0xd9347ae8 03:24:55.825] tp_mqtt INFO IN: [POLL#10] [io_iot_hub_umqtt_connection_on_receive:446]
[Pid=1:Tid=0xd9347ae8 03:24:55.826] tp_mqtt INFO OUT: [DATA#5] [io_iot_hub_umqtt_connection_on_send:324]
[Pid=1:Tid=0xd9347ae8 03:24:56.202] tp_mqtt INFO IN: [POLL#11] [io_iot_hub_umqtt_connection_on_receive:446]
[Pid=1:Tid=0xd9347ae8 03:24:56.202] tp_mqtt INFO OUT: [DATA#7] [io_iot_hub_umqtt_connection_on_send:324]
[Pid=1:Tid=0xd9347ae8 03:24:56.621] tp_mqtt INFO IN: [POLL#12] [io_iot_hub_umqtt_connection_on_receive:446]
[Pid=1:Tid=0xd9347ae8 03:24:56.652] tp_mqtt INFO IN: [DATA#13] [io_iot_hub_umqtt_connection_on_receive:446]
[Pid=1:Tid=0xd9347ae8 03:24:56.656] tp_mqtt INFO OUT: [POLL#13] [io_iot_hub_umqtt_connection_on_send:324]
[Pid=1:Tid=0xd9360ae8 03:24:56.679] server_sk NOTICE Remote close received (s: 1, 0x5631c78ca420) [prx_server_socket_on_end_receive:868]
[Pid=1:Tid=0xd9360ae8 03:24:56.679] server_sk NOTICE Remote close received (s: 2, 0x5631c78ca420) [prx_server_socket_on_end_receive:868]
[Pid=1:Tid=0xd9347ae8 03:24:56.679] tp_mqtt INFO OUT: [DATA#9] [io_iot_hub_umqtt_connection_on_send:324]
[Pid=1:Tid=0xd9347ae8 03:24:56.967] tp_mqtt INFO IN: [POLL#14] [io_iot_hub_umqtt_connection_on_receive:446]
[Pid=1:Tid=0xd9347ae8 03:24:56.967] tp_mqtt INFO OUT: [POLL#14] [io_iot_hub_umqtt_connection_on_send:324]
[Pid=1:Tid=0xd9347ae8 03:25:01.415] tp_mqtt INFO IN: [POLL#15] [io_iot_hub_umqtt_connection_on_receive:446]
[Pid=1:Tid=0xd9347ae8 03:25:01.415] tp_mqtt INFO OUT: [POLL#15] [io_iot_hub_umqtt_connection_on_send:324]
[Pid=1:Tid=0xd9347ae8 03:25:01.902] tp_mqtt NOTICE IN: [CLOSE#16] [io_iot_hub_umqtt_connection_on_receive:451]
[Pid=1:Tid=0xd9347ae8 03:25:01.902] tp_mqtt NOTICE OUT: [CLOSE#16] [io_iot_hub_umqtt_connection_on_send:329]
[Pid=1:Tid=0xd9347ae8 03:25:02.434] tp_mqtt NOTICE IN: [PING#18] [io_iot_hub_umqtt_connection_on_receive:451]
[Pid=1:Tid=0xd9347ae8 03:25:02.434] root NOTICE Resolving moxa-mc1121-2821:51210 (family: 0)... [pal_getaddrinfo:1180]
[Pid=1:Tid=0xd9347ae8 03:25:02.438] root INFO   ... moxa-mc1121-2821:51210 -> 10.168.110.35:51210 (moxa-mc1121-2821.guest.corp.microsoft.com) [pal_net_trace_resolved:36]
[Pid=1:Tid=0xd9347ae8 03:25:02.438] root INFO   ... moxa-mc1121-2821:51210 -> 10.168.110.35:51210 (moxa-mc1121-2821.guest.corp.microsoft.com) [pal_net_trace_resolved:36]
[Pid=1:Tid=0xd9347ae8 03:25:02.438] tp_mqtt NOTICE OUT: [PING#18] [io_iot_hub_umqtt_connection_on_send:329]
[Pid=1:Tid=0xd9347ae8 03:25:02.918] tp_mqtt NOTICE IN: [LINK#19] [io_iot_hub_umqtt_connection_on_receive:451]
[Pid=1:Tid=0xd9347ae8 03:25:02.918] pal_sk NOTICE Resolving moxa-mc1121-2821:51210... [pal_socket_open_by_name:1016]
[Pid=1:Tid=0xd9347ae8 03:25:02.919] root NOTICE Resolving moxa-mc1121-2821:51210 (family: 2)... [pal_getaddrinfo:1180]
[Pid=1:Tid=0xd9347ae8 03:25:02.920] root INFO   ... moxa-mc1121-2821:51210 -> 10.168.110.35:51210 (moxa-mc1121-2821.guest.corp.microsoft.com) [pal_net_trace_resolved:36]
[Pid=1:Tid=0xd9347ae8 03:25:02.920] root INFO   ... moxa-mc1121-2821:51210 -> 10.168.110.35:51210 (moxa-mc1121-2821.guest.corp.microsoft.com) [pal_net_trace_resolved:36]
[Pid=1:Tid=0xd9347ae8 03:25:02.920] pal_sk INFO Socket created ... (fd:17) [pal_socket_try_open:865]
[Pid=1:Tid=0xd9347ae8 03:25:02.920] pal_sk INFO Socket connecting ... (fd:17) [pal_socket_connect:723]
[Pid=1:Tid=0xd9347ae8 03:25:23.932] root NOTICE A OS operation resulted in error 111 (Connection refused) [pal_os_last_error_as_prx_error:528]
[Pid=1:Tid=0xd9347ae8 03:25:23.932] pal_sk ERROR Failed connecting socket (er_refused) [pal_socket_connect:749]
[Pid=1:Tid=0xd9347ae8 03:25:23.932] pal_sk ERROR Failed opening socket (er_refused)! [pal_socket_try_open:914]
[Pid=1:Tid=0xd9347ae8 03:25:23.932] pal_sk INFO Socket created ... (fd:17) [pal_socket_try_open:865]
[Pid=1:Tid=0xd9347ae8 03:25:23.932] pal_sk INFO Socket connecting ... (fd:17) [pal_socket_connect:723]

Thanks!

marcschier, 2017-08-16:
Hi, seems the initial discovery succeeded, but then the OPC server returned moxa-mc1121-2821 host as endpoint to connect to.  And I presume given the 10.xx address that name is on the wired subnet and the server there has firewall up, or an old DNS entry, thus er_refused.

If the OPC machine  is moxa-mc1121-2821 then did you move it from one network to another?  And is it possible that DNS still returns the old ip?  Otherwise, I need to look into the OPC reference code to see why a different host name than gethostname would be returned during discovery...

marcschier, 2017-08-16:
This appears to be the same issue as #74.  Investigating.

marcschier, 2017-08-16:
Rather than using HEAD, could you try tag 1.0.3? This could be a regression in the OPC stack introduced when switching to the official nuget. But that's just a hunch.

606simps, 2017-08-17:
Hi, I figured it's a DNS problem. I got it working by adding one entry "192.168.1.100 moxa-mc1121-2821" to the container's /etc/hosts. It is possible that the OPC machine was connected to other network as you mentioned. At the end, adding the server entry to /etc/hosts seems reasonable in the real world.
Thanks!

marcschier, 2017-08-17:
Great.  We added mdns browsing and resolution to the proxy as well (Experimental, see browser.dll sample app), that would allow resolving a server through the broadcasted MDNS service and host record if LDS-ME is used, for technically a DNS-less setup, which is common for OPC-UA environments.

closed 2017-08-17 by marcschier