Azure/iot-edge-opc-proxy #75: build using docker failed

lcao8, 2017-08-17:
I am getting the following error when trying to do a build using docker on centos 7.

docker build -t proxyd https://github.com/Azure/iot-edge-opc-proxy.git
Sending build context to Docker daemon  7.076MB
Step 1/6 : FROM alpine:3.6
 ---> 7328f6f8b418
Step 2/6 : RUN set -ex         &&     apk add --update --no-cache --virtual .build-deps         bash         cmake         build-base         gcc         abuild         binutils         make         linux-headers         libc-dev         libressl-dev         curl-dev         avahi-dev
 ---> Using cache
 ---> ba3522964f56
Step 3/6 : ADD / /proxy_build
 ---> 738cbcde3904
Removing intermediate container 57009353be0a
Step 4/6 : RUN set -ex         &&       bash /proxy_build/bld/build.sh -C Release --skip-unittests --use-zlog --use-dnssd embedded         &&     mv /proxy_build/build/cmake/Release/bin/* /usr/bin         &&     mv /proxy_build/build/cmake/Release/libwebsockets/lib/* /usr/lib         &&     mv /proxy_build/build/cmake/Release/lib/* /usr/lib         &&     rm -rf /proxy_build
 ---> Running in 17e3cad1c09d
+ bash /proxy_build/bld/build.sh -C Release --skip-unittests --use-zlog --use-dnssd embedded
Building Release...
/proxy_build/bld/build.sh: line 132: [: ==: unary operator expected
0
... Success!
+ mv /proxy_build/build/cmake/Release/bin/* /usr/bin
mv: can't rename '/proxy_build/build/cmake/Release/bin/*': No such file or directory
The command '/bin/sh -c set -ex         &&      bash /proxy_build/bld/build.sh -C Release --skip-unittests --use-zlog --use-dnssd embedded         &&     mv /proxy_build/build/cmake/Release/bin/* /usr/bin         &&     mv /proxy_build/build/cmake/Release/libwebsockets/lib/* /usr/lib         &&     mv /proxy_build/build/cmake/Release/lib/* /usr/lib         &&     rm -rf /proxy_build' returned a non-zero code: 1

marcschier, 2017-08-17:
#76 should fix this I believe.

marcschier, 2017-08-17:
Dockerhub also now emitted a successful build again.

closed 2017-08-17 by marcschier

lcao8, 2017-08-17:
build now goes further, but still fails,

Sending build context to Docker daemon  7.102MB
Step 1/6 : FROM alpine:3.6
 ---> 7328f6f8b418
Step 2/6 : RUN set -ex         &&     apk add --update --no-cache --virtual .build-deps         bash         cmake         build-base         gcc         abuild         binutils         make         linux-headers         libc-dev         libressl-dev         curl-dev         avahi-dev
 ---> Using cache
 ---> ba3522964f56
Step 3/6 : ADD / /proxy_build
 ---> Using cache
 ---> 15c655b49091
Step 4/6 : RUN set -ex         &&       bash /proxy_build/bld/build.sh -C Release --skip-unittests --use-zlog --use-dnssd embedded         &&     mv /proxy_build/build/cmake/Release/bin/* /usr/bin         &&     mv /proxy_build/build/cmake/Release/libwebsockets/lib/* /usr/lib         &&     mv /proxy_build/build/cmake/Release/lib/* /usr/lib         &&     rm -rf /proxy_build
 ---> Running in fba56477f0a0
+ bash /proxy_build/bld/build.sh -C Release --skip-unittests --use-zlog --use-dnssd embedded
Building Release...
Building native...
    Release...
-- The C compiler identification is GNU 6.3.0
-- The CXX compiler identification is GNU 6.3.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - found
-- Found Threads: TRUE
-- Found CURL: /usr/lib/libcurl.so (found version "7.55.0")
-- Looking for getopt.h
-- Looking for getopt.h - found
-- Could NOT find Git (missing:  GIT_EXECUTABLE)
-- Found OpenSSL: /usr/lib/libssl.so (found version "2.0.0")
-- Found PkgConfig: /usr/bin/pkg-config (found version "1.3.7")
-- Looking for linux/random.h
-- Looking for linux/random.h - found
--   linking with urandom rand
-- Checking for module 'libsecret-1'
--   Package 'libsecret-1', required by 'virtual:world', not found
-- Checking for module 'gnome-keyring-1'
--   Package 'gnome-keyring-1', required by 'virtual:world', not found
CMake Warning at bld/src/libproxy/CMakeLists.txt:117 (message):
  no secret store - install libsecret-1!

-- Checking for module 'avahi-core'
--   Found avahi-core, version 0.6.32
-- Checking for module 'avahi-client'
--   Found avahi-client, version 0.6.32
--   linking with avahi-core
--   linking with lws
CMake Error at bld/src/libproxy/CMakeLists.txt:213 (add_subdirectory):
  The source directory

    /proxy_build/deps/lws

  does not contain a CMakeLists.txt file.

--   linking with openssl
-- Configuring incomplete, errors occurred!
See also "/proxy_build/build/cmake/Release/CMakeFiles/CMakeOutput.log".
The command '/bin/sh -c set -ex         &&      bash /proxy_build/bld/build.sh -C Release --skip-unittests --use-zlog --use-dnssd embedded         &&     mv /proxy_build/build/cmake/Release/bin/* /usr/bin         &&     mv /proxy_build/build/cmake/Release/libwebsockets/lib/* /usr/lib         &&     mv /proxy_build/build/cmake/Release/lib/* /usr/lib         &&     rm -rf /proxy_build' returned a non-zero code: 1

marcschier, 2017-08-17:
We had this issue before.  Apparently docker in some older versions does not clone recursive.  The solution is to use an official latest docker build or git clone the repo locally using --recursive and then run 'docker build -t proxyd .' in the repo root.