Azure/iot-edge-opc-proxy #77: Failed to create iothub registry

lcao8, 2017-08-18:
I did a docker build on the latest and when I ran it against an IoT Hub in the azure cloud, I am getting the following error and proxyd terminated. I am running it on centos 7. I don't see the error if I run older builds from docker hub.

docker run -it --network=host proxyd -c "HostName=sch-cf805a3.azure-devices.net;SharedAccessKeyName=iothubowner;SharedAccessKey=****"

=== Azure iot-edge-reverse-proxy 1.0.3. ===

[Pid=1:Tid=0xf543db48 15:30:36.548] root NOTICE Pal does not provide credential storage support. Ensure you properly safeguard all secrets on this device! [pal_cred_init:21]
ERROR: Failed to create iothub registry from --connection-string supplied connection string.

Operation failed.

JuergenKosel, 2017-08-18:
You need to avoid, that the shell interprets the special characters in the connection string!
Use ' instead of "

lcao8, 2017-08-18:
Thanks for the quick response!

I changed " around the connection string to ', still got the same error.

docker run -it --network=host proxyd -c 'HostName=sch-cf805a3.azure-devices.net;SharedAccessKeyName=iothubowner;SharedAccessKey=****'

marcschier, 2017-08-19:
Were you able to debug this?  If not I suggest using 1.0.2 for now.  It is what we use in our connected factory simulation.

lcao8, 2017-08-22:
I haven't got a chance to debug this. I switched to 1.0.2 for the azure connected factory. It does work much better. Thanks!

marcschier, 2018-02-01:
I reproduced this with ", or ' quoting on windows.  I fixed this by trimming quotes from the connection string passed through command line.  Fix will be in 1.0.4

gasherma, 2018-06-07:
This is fixed in the latest version. If you still see the issue, please reopen it.

closed 2018-06-07 by gasherma