Azure/iot-edge-opc-proxy #82: Registration device fails!!

JORGEGO, 2018-01-24:
Until last week all was correctly, but now when I update from Microsoft.Azure.Devices 1.3.2 to 15.0  add new device every time same Exception:

{"Message":"ErrorCode:ArgumentNull;BadRequest","ExceptionMessage":"Tracking ID:8f81a11b2ba747dfaeecb162605e33f5-G:6-TimeStamp:01/23/2018 20:39:04"}

For your help.

marcschier, 2018-01-25:
@JORGEGO, can you provide more details, context, ideally repro instructions - i.e. how you got this error?  I assume some JSON sent to the IoT Hub endpoint being malformed.  However, iot-edge-opc-proxy does not use the Microsoft.Azure.Devices assembly itself, so this might be occurring in another part of your stack.  Thx.

marcschier, 2018-01-31:
Closing since I did not get any repro.  Please reopen if needed.

closed 2018-01-31 by marcschier

SaurabhRaoot, 2018-03-21:
AddDeviceAsync method of registry throw exception ArgumentNull.

Exception : {"Message":"ErrorCode:ArgumentNull;BadRequest","ExceptionMessage":"Tracking ID:adf7e83e7db046969086702500cbe73b-G:2-TimeStamp:03/21/2018 14:09:50"} Method: ProjectXXX.NTB.FN.DeviceRegistration.IdentityCreationService.CreateDeviceIdentity() Description: Unexpected exception

Parameter, registryManager and output parameter nothing is null but it still throw exception.

Is it issue related to Microsoft.Azure.Devices library ??

Code

	RegistryManager registryManager = null;
	
	void Initialize()
	{
		registryManager = RegistryManager.CreateFromConnectionString(AppSetting.IoTHubConnectionString);
	}

	public async Task<string> CreateDeviceIdentity(string deviceId, string deviceKey)
	{
		Device device = new Device(deviceId);
		Device newdevice =  new Device();

		Exception exception = null;
		bool bExceptionHasOccured = false;

		string token = string.Empty;
		string primarySasToken = Base64Encode(deviceKey);
		string secondarySasToken = Base64Encode($"{deviceId}-{deviceId}");
		device.Authentication = new AuthenticationMechanism
		{
			SymmetricKey = new SymmetricKey
			{
				PrimaryKey = primarySasToken,
				SecondaryKey = secondarySasToken
			}
		};
		try
		{
			newdevice = await registryManager.AddDeviceAsync(device);
			break;
		}
		catch (DeviceAlreadyExistsException)
		{
			token = GetDeviceToken(deviceId);
			break;
		}
		catch (IotHubThrottledException e)
		{
		   
		}
		catch (SocketException e)
		{
		  
		}
		catch (Exception e)
		{
		   
		}
	}
	

Stack Trace
System.ArgumentException:
   at Microsoft.Azure.Devices.HttpClientHelper+<ExecuteAsync>d__36.MoveNext (Microsoft.Azure.Devices, Version=1.0.0.0, Culture=neutral, PublicKeyToken=31bf3856ad364e35)
   at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess (mscorlib, Version=4.0.0.0, Culture=neutral, PublicKeyToken=b77a5c561934e089)
   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification (mscorlib, Version=4.0.0.0, Culture=neutral, PublicKeyToken=b77a5c561934e089)
   at Microsoft.Azure.Devices.HttpClientHelper+<PutAsync>d__12`1.MoveNext (Microsoft.Azure.Devices, Version=1.0.0.0, Culture=neutral, PublicKeyToken=31bf3856ad364e35)
   at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess (mscorlib, Version=4.0.0.0, Culture=neutral, PublicKeyToken=b77a5c561934e089)
   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification (mscorlib, Version=4.0.0.0, Culture=neutral, PublicKeyToken=b77a5c561934e089)
   at System.Runtime.CompilerServices.TaskAwaiter`1.GetResult (mscorlib, Version=4.0.0.0, Culture=neutral, PublicKeyToken=b77a5c561934e089)
   at ProjectXXX.NTB.FN.DeviceRegistration.IdentityCreationService+<CreateDeviceIdentity>d__6.MoveNext (ProjectXXX.NTB.FN.DeviceRegistration, Version=1.0.0.0, Culture=neutral, PublicKeyToken=nullProjectXXX.NTB.FN.DeviceRegistration, Version=1.0.0.0, Culture=neutral, PublicKeyToken=null: D:\ProjectXXX\ProjectXXX.NTB\ProjectXXX.NTB\DeviceRegistrationLib\ProjectXXX.NTB.FN.DeviceRegistration\IdentityCreationService.csProjectXXX.NTB.FN.DeviceRegistration, Version=1.0.0.0, Culture=neutral, PublicKeyToken=null: 122)