Azure/iot-edge-opc-proxy #86: How to deploy this proxy to IoT Edge

Gimly, 2018-07-12:
Hi,

I'm really confused about the documentation in the README. I don't understand how this proxy is supposed to be deployed in IoT Edge. The examples only explain how to run it through Docker but doesn't explain how to deploy it as an IoT Edge module.

Is there something obvious I'm missing? In the [ConnectedFactory sample](https://github.com/Azure/azure-iot-connected-factory) there doesn't seem to be any usage of the IoT Edge device, both the proxy and the OPC UA Publisher are run directly in docker.

Thanks!

eoursel, 2018-07-28:
I am also confused and i don't really understand  the architecture. What is the communication protocol used between this module and the cloud ? MQTT or Websocket? Which Azure service are you using on the Azure side? IOTHUB or Azure service Bus or ...? Thanks

marcschier, 2018-07-28:
Instructions have been updated as to how to run inside IoT edge v2 (main readme.md).

We have no plans to upgrade the Connected Factory simulation to use iot edge v2 at this point.  However, both publisher and proxy have been tested in the IoT edge v2 context, so can be used on real edge gateway devices. 

The proxy predates IoT Edge v2.  The instructions show how to run it in the context of IOT Edge v2, but it will use its own outbound MQTT (web-)socket connection, and will not route through edgeHub.  This makes sense since it is a stream relay, so the data is unintelligible to other modules.  

@eoursel, IoT edge uses a MQTT over websocket connection into Azure IoT Hub, which is the only service needed to use it.

closed 2018-07-28 by marcschier