Azure/iot-edge-opc-proxy #88: Proxy refused create connections

ryan-ms, 2018-08-10:
Hi

I just tested the Azure Iot solution: connected factory, and publisher , proxy component.
I followed the step on the [publisher github repository](https://github.com/Azure/iot-edge-opc-publisher) and [proxy github repository](https://github.com/Azure/iot-edge-opc-proxy) deploy the publisher and proxy as Azure iot edge module.
And used Connected Factory to enable gateway and connect to OPCUA server.

So, after the iot edge start,  I tried to connect to gateway(my laptop) by filling **opc.tcp://tw-rychen-175:62222** in the Connect your own OPC UA server page on Connected Factory web app.

But I got the error from proxy logs as below..
[image]

But I had tested it successfully by using the self container, not sure why it can not work by using the component as iot edge module. 

------

My test environment as below: 

OPC Foundation Sample Server Simulator and OPC Proxy and OPC publisher as iot edge module in the same laptop.

The IP of my laptop is : 10.86.124.71.
In the etc/host file of my laptop, I also added the entry **tw-rychen-175:10.86.124.71**

hansgschossmann, 2018-09-20:
Looks you are trying to connect to the OPC UA server integrated into the OPC Publisher (which is on port 62222). Not sure where you have it running, but you need to ensure that the there is a OPC UA server listening on 172.19.0.2:62222. A simple issue might be you need to export the port 62222, if OPC Publisher is running in a container as well.

ryan-ms, 2018-10-31:
Thanks for replying.

Below are some updated===

My environment,

OS: win10
Running  Publisher and Proxy in Azure iot edge module in my laptop.
Running OPC UA sample server in the same laptop. 
Laptop's hostname and IP  :  tw-rychen-175:10.86.124.91

[image]

Now I can successfully connect to `opc.tcp://publisher:62222`
But when I want to connect to my opc ua server on `opc.tcp://tw-rychen-175:51210/UA/SampleServer ` from connected factory.
[image]

It still show the error message below.
[image]

But When I change the url to `opc.tcp://10.86.124.91:51210/UA/SampleServer`. (use IP instead of hostname)
It can successfully connect to my opc ua server.

[image]

So in the error message

> tw-rychen-175:51210 -> 172.19.0.2:51210

it seems that the host name mapping to IP `172.19.0.2`
But my IP is `10.86.124.91`. Not sure why it map to the wrong IP.
But if I run the publisher and proxy by docker container directly instead of running as iot edge module. it works fine.

Could you kindly gives me some suggestion?

asgaralim, 2019-02-15:
As said in connected factory architecture. I created azure iot edge system on linux system. I deployed OPC publisher and OPC proxy containers on to edge systeam. Could you please provide details how to configure opc proxy container  and opc ua client App ( in cloud), so that opc ua client application send command to opc ua server using opc proxy.  How to implement and configure these components

closed 2019-07-02 by marcschier