Azure/iot-edge-opc-proxy #91: ERROR Failure in HTTP communication: server reply code is 400 [HTTPAPI_ExecuteRequest:594]

sjai013, 2019-03-19:
When trying to run the docker image, the installation fails with the error:
`=== Azure iot-edge-reverse-proxy 1.0.4. ===

[Pid=1:Tid=0x5b0e5b88 01:32:12.878] root NOTICE Pal does not provide credential storage support. Ensure you properly safeguard all secrets on this device! [pal_cred_init:21]
[Pid=1:Tid=0x5b0e5b88 01:32:15.466] xlog ERROR Failure in HTTP communication: server reply code is 400 [HTTPAPI_ExecuteRequest:594]
[Pid=1:Tid=0x5b0e5b88 01:32:15.466] xlog NOTICE HTTP Response:{"Message":"ErrorCode:GenericBadRequest;BadRequest","ExceptionMessage":"Tracking ID:423840d9bdc84fbb8d174f0454f6ef61-G:8-TimeStamp:03/19/2019 01:32:15"} [HTTPAPI_ExecuteRequest:596]
[Pid=1:Tid=0x5b0e5b88 01:32:15.466] xlog NOTICE HTTP Response:{"Message":"ErrorCode:GenericBadRequest;BadRequest","ExceptionMessage":"Tracking ID:423840d9bdc84fbb8d174f0454f6ef61-G:8-TimeStamp:03/19/2019 01:32:15"} [HTTPAPI_ExecuteRequest:596]
ERROR: er_invalid_format Install failed! Check parameters...`

I am running Docker v2.0.0.3 on Windows 10 Build 17134.  I am running the command:
`docker run -it --rm -v D:/shared:/mapped --network iot_edge --name proxy microsoft/iot-edge-opc-proxy:1.0.4 -i -c "[Hidden]" -D /mapped/cs.db`
as described in the Azure Conencted Factory How-To.  

Previous issues describe this as temporary.  I have been trying this for a few hours with the same result.

marcschier, 2019-03-19:
I tried the same image on 10.0.17763.379 with 2.0.0.3, and with and without -i  (but omitted --network) and it worked.   Is it possible that the [Hidden] is a bad connection string?   Or that there is an issue with the --network you are using?

sjai013, 2019-03-19:
Oddly enough, I ran with the -T switch, and then it worked.  Don't know if this was an intermittent thing, or if the -T fixed it.

closed 2019-07-02 by marcschier